betaflight/betaflight-configurator #1367: 10.5.1- LED Strip Function State always shows "None"

Lord-Quake, 2019-04-12:
I have my LED State set at "Function" to "Arm State".
GREEN = unarmed
RED = armed
Every time I load up the configurator "Function" shows "None".
"Save" was done.
On the quad the LEDs function as intended.

Is this a bug in the configurator or is this by design or is there a specific issue I'm having?

mikeller, 2019-04-14:
@Lord-Quake: Sounds like a bug. Can you specify a bit more of your configuration, configurator / firmware versions, CLI `diff`, hardware used?

Lord-Quake, 2019-04-14:
Sure.

It's a HGLRC XJB-145 stock with two LEDs.
I'm running BF 4.0 and Configurator 10.5.1.
Flashed with full erase and I configured the quad from scratch.
The DIFF: [BTFL_cli_QC_4_20190414_073755.zip](https://github.com/betaflight/betaflight-configurator/files/3077049/BTFL_cli_QC_4_20190414_073755.zip)
Worked fine until I updated to the above mentioned versions.

mikeller, 2019-04-14:
Just tried to reproduce with your configuration - for me this is working fine:

[image]

Are you sure you are actually selecting one of the configured LEDs?

Lord-Quake, 2019-04-14:
I'm really sorry for having you go through my lack of smartness.....
Setting up the LEDs is something you do once and then never really change again. Since this was done a year or two ago I had assumed that the state would show up automatically once the configs are loaded. I didn't realize you needed to select the LED in addition to have the actual setting show up. I shall continue exploring BF 4.0 with more brains.

I'm closing this in shame.

Thank you for the time and work you put into this great project!!

closed 2019-04-14 by Lord-Quake