betaflight/betaflight-configurator #1369: bug configurator 10.5.1 windows MODES

Vincentdt, 2019-04-12:
Hello,

I'm stuck when i go to MODES i can only disconnect + i see nothing except ARM see the screen shot.
[image]

McGiverGim, 2019-04-12:
Look here https://github.com/betaflight/betaflight-configurator/issues/1365. Same problem.

mikeller, 2019-04-14:
@Vincentdt: Can you please specify what versions of configurator / firmware and what hardware you are using?

oolcoo, 2019-04-15:
> @Vincentdt: Can you please specify what versions of configurator / firmware and what hardware you are using?

same problem here with my trashcan

Configurator: 10.5.1 (10.5.0 same bug)
Firmware: BTFL 4.0.0
Target: CRAZYBEEF4FR

McGiverGim, 2019-04-15:
Exactly. The solution is to update to Betaflight 4.0 final from the Configurator.

mikeller, 2019-04-16:
Duplicate of #1362.

closed 2019-04-16 by mikeller

Winky1st, 2019-05-07:
Same here...but for a mobula7 with f3 board to update version is mostly impossible as wont work smooth unless loaded the stock 4.0 provided by them and than cant be loaded with configuratr 10.5.1 as per auxiliarttab issue
apologise for my not native English.

Arvenz200, 2020-01-28:
I am also having this same problem with the Modes tab with my new Trashcan. Luckily i had a backup copy i setup months ago when it was workign fine on my computer. Only i want to check what my settings are and lower Pids? So please help me with this? i have the latest Baseflight version. :(