betaflight/betaflight-configurator #1371: Betaflight Failure to open

dcburkhart, 2019-04-13:
I have checked all my drivers.
Stand-alone configurator quit working for me some time ago, but 10.4.0 chrome app still worked so I just uninstalled the stand-alone and continued to use Chrome. Well, I can't update the chrome app to 10.5.1 and would like to flash/configure 4.0.
I have attempted to uninstall and reinstall a second time. I have restarted my computer. Nothing seems to work and I couldn't find any answers.
I have verified that Java is updated to 8.0 build 201
I have removed all old versions of BFC from pc and redownloaded after turning off all firewalls and realtime scanning. Windows firewalls are turned off. I am still getting the same result.

There are no error codes or logs that I can attach.

I have attached a screenshot.
[image]

mikeller, 2019-04-14:
@dcburkhart: 10.5.1 has been released to the Chrome Web Store by now. Nonetheless you should try to fix your drivers and make the standalone version work (there are many sets of instructions on how to fix your drivers on the net) - one day Google will drop support for Chrome Web apps for non-Chromebooks.

dcburkhart, 2019-04-14:
Thanks!

Some peeps helped me out.
It was a corrupt appdata file.
Issue has been resolved.

dcburkhart, 2019-04-14:
I'll shoot over to github and relay the message.

MrBillBill, 2019-04-14:
@dcburkhart, I've been having the same issue.
Could you tell me how you fixed it?

dcburkhart, 2019-04-14:
Might be corrupt personal files. If allowing it through the firewall
doesn't help, try opening C:\Users\YourProfile with Show hidden
files/folders toggled on in Explorer. Look through all of folders
immediately under AppData for anything BetaFlight and rename it, then try
launching again.

MrBillBill, 2019-04-14:
Thanks, I'll give it a try.
I've been having this problem since 10.4 and would like to get away from using the chrome app

dcburkhart, 2019-04-14:
Weird! It was about 5 months ago it broke for me. Wonder if it was a
Windows update or something.
You on Windows 7?

MrBillBill, 2019-04-14:
I wouldn't be surprised, but i'm on Windows 10. 
It's been about that long for me too. I couldn't find a solution at that point, but someone mentioned the chrome app was still working so I went that route.
I want to get back to the standalone as it was working for me before that.
I have some hope to get it fixed now!

MrBillBill, 2019-04-14:
It is back! This is awesome, its been so long!
Thanks for the help

closed 2019-05-21 by stale[bot]