betaflight/betaflight-configurator #1375: Custom mmix calculator for non standard frames

mmakam2, 2019-04-14:
Requesting a feature in configurator/firmware to set motor distances and the mmix automatically be calculated. I recently built a catalyst raging droner, which has a deadcat layout. Setting up the mmix made a significant difference in how the craft handled in turns mainly. Much more locked in and pid numbers are closer as well.

Calculating manually can be a bit of a pain, and the one calculator that is available on the web does not work very well and is buggy. It would be great if it was built right into the platform. There is even a benefit for stretch x or compressed x frames.

apoc, 2019-05-19:
There is already some mmix custom calculator online: https://www.iforce2d.net/mixercalc/

mmakam2, 2019-05-19:
I force calculator works extremely poorly if you have ever tried it

mmakam2, 2019-05-19:
I guess I'm the only one who actually mixes motors? 😂 It seems to get me a good result when I do it, pid controller knows exactly what force each motor will add to the bird.

apoc, 2019-05-19:
I tried it with my squashed X frame and it worked good, I did not try any exotic one :)

closed 2019-06-25 by stale[bot]