betaflight/betaflight-configurator #1383: OSD display dislplay_name

skulldrones, 2019-04-14:


skulldrones, 2019-04-14:
The variable name of aircraft is 'name', but in the OSD, the system displays the info of variable 'display_name'. They are different variables and the previous version of Betaflight them are the same.
GUI: Betaflight 10.5.1
Firmware: FURYF4OSD v. 4.0.0

mikeller, 2019-04-14:
@skulldrones: With Betaflight 4.0 there are now 2 different, independent fields for naming:

- `name` used in OSD, configurator, and for log / config files;
- `display_name` used only in OSD.

skulldrones, 2019-04-14:
Okay, but I only assign value to variable name and it did not display the value in OSD ... it was only displayed: `display_name -`

skulldrones, 2019-04-14:
# dump all

# version
# Betaflight / FURYF4OSD (FY4O) 4.0.0 Apr 11 2019 / 07:16:52 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 
mcu_id 0021002b354b500420343246
signature 

**# name
name Skull 249HD**

# resources
resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource MOTOR 5 NONE
...
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

**# display_name
display_name -**

# master
set gyro_hardware_lpf = NORMAL
set gyro_sync_denom = 1
set gyro_lowpass_type = BIQUAD

mikeller, 2019-04-15:
@skulldrones: Looks like you enabled the wrong element then:

[image]

closed 2019-05-22 by stale[bot]