betaflight/betaflight-configurator #1387: Error on displaying korean language at "firmware update"

elin-neo, 2019-04-15:
It is OK when the application is started.
[image]

But, if you click on the firmware update icon, the string is changed as shown below.
[image]

I tested on windows10(x64).

McGiverGim, 2019-04-15:
I will fix it for the next version. Thanks!

mikeller, 2019-04-15:
@elin-neo: We are using Crowdin (https://crowdin.com/project/betaflight-configurator) to maintain the translations for the configurator. Feel free to register on the platform and join the project if you want to help us maintain the Korean translation.

McGiverGim, 2019-04-15:
Looking at Crowdin, it seems that it has been fixed by the proofreader of Korean @fixmania, so it will be ok in the next version.

elin-neo, 2019-04-16:
It has same problem with latest version of Crowdin translation.

- In the file /locales/ko/messages.json :
```javascript
"flashTab": {
        "message": "펌웨어<\/br>업데이트"
    }
```
This version has problem that I reported.

- I tested after change like this :
```javascript
"flashTab": {
        "message": "펌웨어<br>업데이트"
    }
```
 This has same problem.

- I test with old version :
```javascript
"flashTab": {
        "message": "펌웨어 업데이트"
    }
```
 This has no problem after click the "update firmware" icon.
But, it looks ugly.  
[image]
I think that is the reason @fixmania edited the string as `"펌웨어<br>업데이트"`.

- Anyhow, the problem is the string is changed after click the "update firmware" icon. I think `<br>` makes some effect to display string.

mikeller, 2019-04-16:
@elin-neo: There is not much point in reporting translation problems in here, as the translators do not follow GitHub. Instead, please create an account on Crowdin, join https://crowdin.com/project/betaflight-configurator, and report the problems there.

fixmania, 2019-04-16:
I know this problem that is only in translation. Crowdin has not solved
this problem.

Japanese language is also inserted <br> code.

But as @elin-neo mentioned, after click the button it looks bad.

So I translated it to "업데이트 펌웨어" now.

Next version it looks nice than before.

elin-neo, 2019-04-16:
Ok I see. 
I reported here because I think this is not the problem of translation.

closed 2019-04-16 by elin-neo

mikeller, 2019-04-16:
@elin-neo: The fact that the spurious html tag is only shown in Korean and Japanese suggests that the problem is in the language files for these languages - otherwise the problem would exist for other languages as well.

fixmania, 2019-04-16:
@elin-neo @McGiverGim @mikeller I will check it this night. This problem seems to have caused an error in tag recognition as html tag was converted to xml format after clicking the button. The original English text does not contain a tag, so it doesn't matter, but since Korean and Japanese are different in word order, need to add tag due to the limited number of characters, but I will test it today. If this is not resolved, the word order is different, but there is no problem with the meaning transmission, so I will translate it according to the number of characters.

elin-neo, 2019-04-16:
I have tested with the string "업데이트 펌웨어", and It is good. I like this idea. 👍

elin-neo, 2019-04-16:
I don't know java script language. So I afraid of editing this code, but I tried something. If I remove line 98 from the file src\js\serial_backend.js, this problem is disappeared.
// $('div.open_firmware_flasher a.flash_state').text(i18n.getMessage('flashTab'));

> src\js\serial_backend.js Line 89 :
```
    $('div.open_firmware_flasher a.flash').click(function () {
        if ($('div#flashbutton a.flash_state').hasClass('active') && $('div#flashbutton a.flash').hasClass('active')) {
            $('div#flashbutton a.flash_state').removeClass('active');
            $('div#flashbutton a.flash').removeClass('active');
            document.getElementById("tab_landing").style.display = "block";
            document.getElementById("tab_help").style.display = "block";
            $('#tabs ul.mode-disconnected .tab_landing a').click();
        } else {
            $('#tabs ul.mode-disconnected .tab_firmware_flasher a').click();
            // $('div.open_firmware_flasher a.flash_state').text(i18n.getMessage('flashTab'));
            $('div#flashbutton a.flash_state').addClass('active');
            $('div#flashbutton a.flash').addClass('active');
            document.getElementById("tab_landing").style.display = "none";
            document.getElementById("tab_help").style.display = "none";
        }
    });
```
I am not sure about this and I am worried about the side effect after changing like this...
Please someone check about this.

McGiverGim, 2019-04-16:
The message is being initialized in the HTML code directly, this is the reason why it works deleting the line:
https://github.com/betaflight/betaflight-configurator/blob/0ef835947f0df7de63336aafd903fa3a26b85592/src/main.html#L134-L139

I think the message does not change never. Maybe at some point it changed and by this reason the `serial_backend.js` modified it again, but it seems I can't find any place in the code where this text changes.

So to me is safer to remove this line. Other solution is to change it:
```
$('div.open_firmware_flasher a.flash_state').text(i18n.getMessage('flashTab'));
```
by
```
$('div.open_firmware_flasher a.flash_state').html(i18n.getMessage('flashTab'));
```
The html will not escape the < and > characters and the `<br>` will be a carriage return.

elin-neo, 2019-04-16:
Thank you for kind explanation, @McGiverGim.

fixmania, 2019-04-16:
@elin-neo @McGiverGim Yeah, I tried to add tag <br>, <br />, <\n>, <\t>, <p></p>, etc. but not solved. 
@McGiverGim has the answer about it. 

When the flash button is active(yellow color), the string of it is shown on html content. 
But when you click the button, it should call the string by java script. 

I have tested to change serial_backend.js '.text()' to '.html()', it was shown correctly without <> tags. 

If @McGiverGim can change the code, it might be the correct translation.

Otherwise, I have a idea. I will translate "Update Firmware" to "FC 펌웨어 업데이트".
This will be shown like the below.
[image]

McGiverGim, 2019-04-16:
I can change the code. Give me some time and I will push an update. It has no sense to change the text for the same text when active.