betaflight/betaflight-configurator #1391: New Configurator has viruses according to Virus Total

Rosinstein, 2019-04-15:
SHA-256 | 91fb0e11fa5963cac9c9beeb084004643a15a8de134b207d5b02c9298a22f6f6
3 | 70
betaflight-configurator-installer_10.5.1_win32.exe
78.25 MB
2019-04-15 00:00:07 UTC
Antiy-AVL

GrayWare/Win32.Presenoker

Trapmine

malicious.moderate.ml.score

VBA32

Adware.Presenoker

Please address this because I'm unable to update to the latest.

mikeller, 2019-04-16:
@Rosinstein: That's a false alarm. Unfortunately virus scanners seem to regularly detect applications built with NW.js (like the configurator is) as viruses.

Can you check if this one isn't detected: https://drive.google.com/open?id=1JcdKsPY22LvTXW3fJFejNaGpZhbFE0xX

Rosinstein, 2019-04-16:
Im a former Cyber security professional
https://www.linkedin.com/in/cesarsalas/ Id like to be as much service as
possible with regards to this anything you need tested, I have a beta75x
tiny whoop, I've also built a f450 flamewheel on a pixhawk board.
Below is what its still testing 8positive for.

*Antiy-AVL*GrayWare/Win32.Presenoker
*Bka*HW32.Packed.
*Trapmine*suspicious.low.ml.score
VBA32Adware.Presenoker

https://www.virustotal.com/#/file-analysis/ZTVlMmEzM2EwYmEyMDEzM2E1MDQ5MzFiYjAzODAxNTg6MTU1NTM3OTkzNQ==

The reason why I opened the ticket and created the account was so that you
could see this error attached from Windows 10.  Windows defender warns you
not to install.  If you can get digitally signed on your installer program
for windows you might be able to get past all of this.  Not sure, but I'm
here to help you.

mikeller, 2019-04-16:
@Rosinstein: Does the second build that I linked in https://github.com/betaflight/betaflight-configurator/issues/1391#issuecomment-483472287 show the same problem?

Rosinstein, 2019-04-16:
> linked in #1391 (comment)

Identically. Let me know or feel free to upload them yourself to
virustotal.com its what cyber security researches use.  I also uploaded to
Kaspersky
https://whitelisting.kaspersky.com/advisor#search/0xE5E2A33A0BA20133A504931BB0380158
they
say its fine.

Docteh, 2019-04-18:
I think for digitally signing the configurator would also need to embed the app into the signed .exe, otherwise the signed exe could be used with a bad app... 

I would like to see CI build for the `dist` gulp task. Then a user could supply their own nwjs executable if they have concerns with the .exe 

I had trouble finding anything on presenor other than vague mention of triggering a signature. Not really good for cyber security, makes it sound like it's not real issue.

Rosinstein, 2019-04-18:
Personally, I just don't want people thinking this is at all intentional or
mistaken when its just a match on heuristics, I know some people will just
be turned off which is why i mentioned it.   I'm still using the previous
chrome version until it clears that is if nothing major comes out the
changes.  Although the OSD improvements i wanted to check out :D

closed 2019-05-25 by stale[bot]