betaflight/betaflight-configurator #1398: LED Profiles missing in 10.5.1

pkruger, 2019-04-20:
BF 4.0.0 includes LED profiles for STATUS, BEACON and RACE however it seems when the Configurator was updated to support BF 4.0.0 LED profiles was missed.  LED Profile selection is available in the adjustments tab, but the LED tab does not have functionality to select the profile, configure the race color or the beacon color and interval.  Can this please be added?

McGiverGim, 2019-04-20:
I don't use LEDs, so I'm not sure what profiles are. Can you try to explain it and suggest what needs to be added to Configurator exactly? I've seen the firmware PR but that says nothing to me.

Docteh, 2019-04-27:
https://github.com/betaflight/betaflight/blob/master/docs/LedStrip.md as usual I read through the code and found the docs at the very end. The configurator would need an item that shows the currently configured race color. 

```
# set color
ledstrip_visual_beeper_color = WHITE
Allowed values: BLACK, WHITE, RED, ORANGE, YELLOW, LIME_GREEN, GREEN, MINT_GREEN, CYAN, LIGHT_BLUE, BLUE, DARK_VIOLET, MAGENTA, DEEP_PINK

ledstrip_race_color = ORANGE
Allowed values: BLACK, WHITE, RED, ORANGE, YELLOW, LIME_GREEN, GREEN, MINT_GREEN, CYAN, LIGHT_BLUE, BLUE, DARK_VIOLET, MAGENTA, DEEP_PINK

ledstrip_beacon_color = WHITE
Allowed values: BLACK, WHITE, RED, ORANGE, YELLOW, LIME_GREEN, GREEN, MINT_GREEN, CYAN, LIGHT_BLUE, BLUE, DARK_VIOLET, MAGENTA, DEEP_PINK
```

I guess a drop down menu of the color choices would do it? The usual standard is to introduce features into the CLI first, and then add to configurator, so maybe the guy who wrote the feature will introduce it to the configurator for the 4.1.0 release.

Hey! add it yourself @pkruger :p

mikeller, 2019-04-27:
@pkruger: Policy is to add new parameters to MSP (and make them available in the configurator GUI) only one release after they have been introduced in CLI. So these new parameters can be added for Betaflight 4.1.

AlexandreBonneau, 2022-12-30:
Please re-open this issue, since that are still no LED profiles in betaflight-configurator.

Ideally, instead of having basic 'one color only' profile, it would be better to use the same kind of logic than for PID and Rates profiles.
For instance I would like on my 3-way switch to have a `ledlow | LED profile #1 | Led profile #2`configuration.
The LED profile would then be as complex as they are now (ie. colors, larsen, orientation, etc. you name it).

haslinghuis, 2022-12-31:
Reading the value from `MSP` is easy. But setting is not. It's just `RACE`, `BEACON` or `STATUS`.

Also don't confuse profiles with PID or RATE profiles having 4 different profiles. This is a single value for a different purpose not implementing multiple LED profiles.

haslinghuis, 2024-05-06:
Closing as outdated

closed 2024-05-06 by haslinghuis