betaflight/betaflight-configurator #1403: OSD tab won't display content

atomgomba, 2019-04-25:
When I connect my OMNIBUSF4FW (Build #1579) to the Configurator (981e2190 master), the contents for OSD tab don't show and I have the following in the console:

```
Firmware is transmitting a different number of statistics (20) to what the configurator is expecting (23)
decode @ osd.js:1613
(anonymous) @ osd.js:1890
r @ bluebird.min.js:31
i._settlePromiseFromHandler @ bluebird.min.js:30
i._settlePromise @ bluebird.min.js:30
i._settlePromise0 @ bluebird.min.js:30
i._settlePromises @ bluebird.min.js:30
r._drainQueue @ bluebird.min.js:29
r._drainQueues @ bluebird.min.js:29
drainQueues @ bluebird.min.js:29
Promise.then (async)
r @ bluebird.min.js:30
r._queueTick @ bluebird.min.js:29
s @ bluebird.min.js:29
p.hasDevTools.r.settlePromises @ bluebird.min.js:29
i._fulfill @ bluebird.min.js:30
i._resolveCallback @ bluebird.min.js:30
(anonymous) @ bluebird.min.js:30
(anonymous) @ msp.js:236
MspHelper.process_data @ MSPHelper.js:1383
(anonymous) @ msp.js:139
notify @ msp.js:138
read @ msp.js:126
read_serial @ serial_backend.js:461
EventImpl.dispatchToListener @ extensions::event_bindings:413
publicClassPrototype.(anonymous function) @ extensions::utils:138
EventImpl.dispatch_ @ extensions::event_bindings:397
dispatchArgs @ extensions::event_bindings:267
dispatchEvent @ extensions::event_bindings:276
bluebird.min.js:29 Unhandled rejection TypeError: Cannot read property 'src' of undefined
    at Object.OSD.generateTimerPreview (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/osd.js:273:42)
    at Object.preview (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/osd.js:848:28)
    at Object.decode (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/osd.js:1707:37)
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/osd.js:1890:29
    at r (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:31:7722)
    at i._settlePromiseFromHandler (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:30:13044)
    at i._settlePromise (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:30:13847)
    at i._settlePromise0 (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:30:14548)
    at i._settlePromises (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:30:15878)
    at r._drainQueue (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:29:3148)
    at r._drainQueues (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:29:3209)
    at drainQueues (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/libraries/bluebird.min.js:29:1238)
D @ bluebird.min.js:29
k @ bluebird.min.js:29
E @ bluebird.min.js:29
e._notifyUnhandledRejection @ bluebird.min.js:29
r._drainQueue @ bluebird.min.js:29
r._drainQueues @ bluebird.min.js:29
drainQueues @ bluebird.min.js:29
Promise.then (async)
r @ bluebird.min.js:30
r._queueTick @ bluebird.min.js:29
s @ bluebird.min.js:29
p.hasDevTools.r.settlePromises @ bluebird.min.js:29
i._fulfill @ bluebird.min.js:30
i._resolveCallback @ bluebird.min.js:30
(anonymous) @ bluebird.min.js:30
(anonymous) @ msp.js:236
MspHelper.process_data @ MSPHelper.js:1383
(anonymous) @ msp.js:139
notify @ msp.js:138
read @ msp.js:126
read_serial @ serial_backend.js:461
EventImpl.dispatchToListener @ extensions::event_bindings:413
publicClassPrototype.(anonymous function) @ extensions::utils:138
EventImpl.dispatch_ @ extensions::event_bindings:397
dispatchArgs @ extensions::event_bindings:267
dispatchEvent @ extensions::event_bindings:276
```

McGiverGim, 2019-04-25:
Seeing the error, it seems some problem with https://github.com/betaflight/betaflight-configurator/pull/1381

I will try to reproduce it later.

McGiverGim, 2019-04-25:
I can confirm the problem, but it does nothing to do with the #1381. I will try to find the culprit.

McGiverGim, 2019-04-25:
Problem found: the Configurator has merged https://github.com/betaflight/betaflight-configurator/pull/1390 but the firmware part was not: https://github.com/betaflight/betaflight/pull/8014

Until merged the firmware part it will crash. :(

atomgomba, 2019-04-25:
Ok, so I just compile the firmware locally to set up my OSD. Thanks for the investigation!

atomgomba, 2019-04-25:
krzysztofmatula:km-stats-postflight has conflicts against latest master

atomgomba, 2019-04-25:
Sorry, I didn't mean to close this. My daughter (1.5yo) smashed my laptop.

mikeller, 2019-04-28:
Sorry, my bad for mergeing one but not the other. Fixed by mergeing betaflight/betaflight#8014.

McGiverGim, 2019-05-17:
As far as I know, this is fixed. I close the issue.

closed 2019-05-17 by McGiverGim