betaflight/betaflight-configurator #1409: DShot beacon RX_SET / RX_LOST swapped?

RipperDrone, 2019-04-30:
Latest Nightlies dated Apr 29 for BF and BFC on Aikon F4: 

RX_SET (switch based) Dshot beacon doesn't sound unless I activate RX_LOST beacon. It seems the 2 beacon features are swapped / mixed up.

RipperDrone, 2019-05-01:
I tried to trace the code in both BF and BFC, but due to lack of understanding I could not really narrow it down:

- RX_SET and RX_LOST seem to be defined consistently, no confusion here
- descriptors seem to be right in BFC's DShot beacon tab as well

Finally, I could not identify where values from the switches tab are being assigned to the RX_SET status flag. Maybe there is a bug in the code assigning the switch value to RX_LOST state rather than RX_SET?

Afraid I would need some help of whoever is more familiar with Java and style sheets... :-/

mikeller, 2019-05-05:
@RipperDrone: Can you close this please?

RipperDrone, 2019-05-08:
sure, sry.

closed 2019-05-08 by RipperDrone