betaflight/betaflight-configurator #1414: In CLI tab 'Save to File' button suggests incorrect filename if there are "bad" symbols in craft_name.

ch3at, 2019-05-02:
Workaround is manually edit file name, which can be frustrated for beginners.
I think, configurator should replace all 'bad' symbols into 'good' one - underscore, for example.

mikeller, 2019-05-02:
@ch3at: The problem is that what is a 'bad' character is not dependend on configurator, but on the operating system it is running on - when running on linux, pretty much all characters will be valid. So unless we go to extreme lengths and detect the operating system (and in some cases the file system used), there is no way to properly solve this - I think what should be done is adding a tooltip to the 'Name' field in configurator, explaining to the user that using special characters in the name may lead to problems.

Docteh, 2019-05-09:
"proper" way would be to figure out a set of character to filter on all OS and see if filtering all those characters cause complaints. I can run down a list for windows and Linux. Maybe someone reading this runs OSX?

On windows I thought it properly complains when you try and hit save. 

What craft_name are you using?

mikeller, 2019-05-09:
I can help you for linux: The list is `\0`. :stuck_out_tongue:

Docteh, 2019-05-13:
`filename = filename + '_' + CONFIG.name.trim().replace(/[<>:"/\\|?* ]/g, '_');`

closed 2019-06-19 by stale[bot]