betaflight/betaflight-configurator #1419: Autocomplete broken on 'set <parameter name>'

mikeller, 2019-05-05:


McGiverGim, 2019-05-05:
I didn't know you can use set in this way. I always used get for this.

mikeller, 2019-05-05:
Yes, you can. And it's a lot more convenient, because this allows you to use the command cache to then `set <parameter name> = <new value>` in the subsequent command.

McGiverGim, 2019-05-07:
@Cleric-K have you time to look at this?

Cleric-K, 2019-05-07:
Hi, I was on the go last few days. I'll take a look at this later today.
Thanks for mentioning my nick, I'm not subscribed to all events in the repos so I was not aware of this issue.

So basically this would be usable only for _non_-lookup values, because for lookup values you immediately get the suggestions and pressing enter will send the first in the list (or would have to press `Esc` first to hide the suggestions).
Is this the desired behavior?

Another possibility would be to _not_ show the lookup suggestions unless `Tab` is pressed or at least one character is typed in. Although I personally enjoy the lookup values popping up immediately.

Yet another variant is just not to add the `=` automatically. But I suspect that not that many users use `set` to get values and thus would prefer the `=` to be added auto.

mikeller, 2019-05-07:
@Cleric-K: Yeah, not sure what best to do here. I'd almost say add the `=` once the parameter name has been selected, but if the user then hits [Enter], strip the `=` again and send `set <parameter name>`, based on the assumption that sending `set <parameter name> =` does not make any sense, and will evaluate to sending 0 (even if this isn't a valid value).

Cleric-K, 2019-05-07:
Yes, that's the clear part. The question is what to do in case of lookup parameters?

mikeller, 2019-05-07:
Heh, even worse there. But I'd essentially go for not selecting one, and just to a `set <parameter name>` if the user doesn't click / type / scroll to select one - pre-selecting the first value does not add much I think.

Cleric-K, 2019-05-07:
I'll take a look but from memory it was not easily doable to make a list with no selected item. The selected index is buried in the lib code and I didn't see provision for index _none_.

In that case maybe the solution would be to show lookup values only after `Tab` or first char in.

Cleric-K, 2019-05-07:
I looked around and here're my thoughts:

* Variant 1 - removing `=` before sending

As we said, one way to go around the problem with the lookup values would be to show the suggestions only if at least a char is typed in or `Tab` is pressed. Unfortunately this has the following caveat: if the suggestion list is _not_ showing, then pressing `Enter` is not handled by the autocomplete lib (it handles events only if the popup is active). So the `Enter` goes straight to the cli handler. This means that if we want to remove the `=` we have to do it there, which makes the whole thing a little ugly.

As a work around we can still show the lookup suggestions but inject an empty first entry, thus pressing enter would be detectable.

Yet, we still have the problem that if `Esc` is pressed (to hide the suggestions) and then press `Enter`, the `=` will be sent verbatim. Thus we have slightly inconsistent behavior depending on whether suggestions are shown or not.

* Variant 2

We can insert additional suggestion step.
After selecting the param _name_ we can show a static two entry list:
`=`
`<show value>`

Selecting the first will continue as in the current implementation. Selecting the second will execute and perform the _get_.
-- The cons of this method is that additional step is added.
-- The pros is that we avoid the inconsistent behavior and the magical erasure of the `=`. As a bonus this will actually hint at users that getting values with the `set` command is actually possible.

Additionally we can implement the `set` command in a similar way as `get` - adding the incomplete input as first entry in the suggestion list. Pressing enter on that would not need to open the `=` step but just execute the getting. I don't know if that will be useful.

mikeller, 2019-05-07:
I actually quite like the second variant. You are correct in that the current implementation simply shortcuts the `set <parameter name> version of the command, so there should be an additional step to add the `=`. Yes, it adds a step, but this is just because the current implementation is missing one. And overall the tab completion still improves the workflow drastically.

McGiverGim, 2019-05-17:
This was fixed in #1424 and https://github.com/betaflight/betaflight-configurator/pull/1436 so I suppose that we can close this issue? Is something pending?

closed 2019-05-18 by mikeller

Asizon, 2020-01-31:
Returning to this situation @mikeller , Why have two functions for the same thing? I think we should standardize this, `Set` command only to change information and not to show values, `Get` command to show information, I think it would be a clearer and more concise way. In this way it could also be shown again `=` in the set autocomplete, and use each command only for one function. Another possibility is to eliminate `Get` and use `Set` for everything?

mikeller, 2020-01-31:
@Asizon: Not sure why you want to 'fix' what is not broken? The `set <value>` syntax is useful for discovery, i.e. you want to set the motor protocol to Dshot, you start typing `set motor`, then you stop because you are not sure what the exact name of the parameter is. So you hit [Enter], and you get

```
# set motor
[...]
motor_pwm_protocol = ONESHOT125
Allowed values: OFF, ONESHOT125, ONESHOT42, MULTISHOT, BRUSHED, DSHOT150, DSHOT300, DSHOT600, PROSHOT1000
[...]
```