betaflight/betaflight-configurator #1422: Betaflight 10.5.1 and CrazybeeF4FS 4.0.0 broke my Eachine Trashcan

norrath93, 2019-05-06:
Hello,
i have a major issue with betaflight and my Eachine Trashcan:

After i flashed the firmware to 4.0.0
, my RSSI is constant under 30% (sometimes it says 101-96)
 and i get the error code Bad_rx_recovery evertime i try to launch.

I flashed back to 3.5 but it still broken and back to 4.0.3 and back to 3.5.6 and so on.
I Resetet my Jumper and done all i know.

When i try to start it just flipps and spinns as crazy for a second before it shutsdown and the message Bad_RX_Recover.
In the Betaflight configurator i get also the Bad_Rx_Recovery every 10sec.

I made a full chip erase and bound it again to my Jumper T8SG V2 with Flysky AFHDS 2A.

But now the weird part, i done that 2 weeks ago with another Trashcan (My First) and it was also broken and unusable.
So i talked with Banngood and they send my a replacement and now 2 Weeks later with my 2.Trashcan the same again……

I did everything but i still cant get a good Signal and start.

It cant be a bad RX because it flow right out of the box with my Jumper T8SG v2 twice so...…… something software based must be the issue.

In the Reciever Tab in Betaflight everything works normal and all Settings are normal.

It is the spi Flysky AFHDS 2A reciever build in the fc.

[image]

mikeller, 2019-05-07:
@norrath93: Have you tried flashing back to the original firmware that is provided by the manufactuer?
Eachine is well known to illegally supply custom modified firmware for their boards without disclosing the sources. It looks like in this case they modified the hardware, causing it to not run reliably with an official build of Betaflight.

norrath93, 2019-05-07:
How to i get the orginal Firmware? 

I wrote to banggood.... now i wait for an reply.

mikeller, 2019-05-07:
@norrath93: Banggood is probably your best chance.

McGiverGim, 2019-05-07:
I have seen sometimes the BAD RX when I power on the quad with the ARM switch enabled. I see in the capture that the ARM switch is enabled too. Do you have the TX on? If true, verify the modes tab, I'm sure something is wrong there.

norrath93, 2019-05-07:
> I have seen sometimes the BAD RX when I power on the quad with the ARM switch enabled. I see in the capture that the ARM switch is enabled too. Do you have the TX on? If true, verify the modes tab, I'm sure something is wrong there.

Everything is correct in the mode tab, arm switch on aux 1 2way switch but like i sayed i flown it before but then this Firmware upgrade...

Docteh, 2019-05-07:
Is the flight controller from happy model?http://www.happymodel.cn/index.php/2019/01/05/crazybee-f4-pro-series-flight-controller-firmware/ or did eachine clone that? 

If the Crazybeef4fs firmware there works, as a customer who bought the item, you could request source code.

norrath93, 2019-05-07:
> Is the flight controller from happy model?http://www.happymodel.cn/index.php/2019/01/05/crazybee-f4-pro-series-flight-controller-firmware/ or did eachine clone that?
> If the Crazybeef4fs firmware there works, as a customer who bought the item, you could request source code.

I flashed it and tried but no hope still broken as hell.
It must be cloned by Eachine then i hope someone has the right firmware here.

mikeller, 2019-05-07:
@norrath93:

> It must be cloned by Eachine then i hope someone has the right firmware here.

As I already stated in https://github.com/betaflight/betaflight-configurator/issues/1422#issuecomment-489829688, the ball in this case here is with Eachine - unless they release firmware and a target for this board, or at least provide the spec for their hardware so others can do the work of creating a target for their hardware, there is nothing anybody else can do.
So instead of pestering people in here, please go and pester Eachine. Or get the money for this hardware refunded and use it to buy hardware from a manufacturer that is not breaking licensing conditions and throwing their customers under the bus.

closed 2019-05-07 by mikeller