betaflight/betaflight-configurator #1423: Installing BFC on Windows after building a release takes forever

serran0, 2019-05-07:
All of the resources in node_modules seem now to be included within installation.
Which means installing a compiled release now takes forever (3+ hours).

[image]

mikeller, 2019-05-08:
@serran0: Can you please supply some more information to help narrow this down, like what version you are trying to build, and if there is an older version that built without this issue?

Docteh, 2019-05-08:
I've been wondering about that. For reproducing the error I just did `git pull` directly followed by  `yarn gulp dist --win32` and when I look in dist/node_modules there is a fair amount of stuff in there.

Installed 10.5.1 configurator node_modules (7 items according to windows explorer)
[image]

dist/node_modules (642 items according to windows explorer)
[image]

I tried specifying --win32 and --linux64 no change for what ends up in dist.

McGiverGim, 2019-05-08:
I have observed the same problem with the Blackbox Viewer. The Blackbox compress the content into a file, so the installer time does not increase but when executing, I suppose it decompress it because it takes forever.
```
var nwBuilderOptions = {
    version: '0.38.0',
    files: './dist/**/*',
    macIcns: './images/bf_icon.icns',
    macPlist: { 'CFBundleDisplayName': 'Betaflight Blackbox Explorer'},
    winIco: './images/bf_icon.ico',
    zip: false
};
```
This `zip`: `true` or `false` makes it compress or not the nwbuilder output.

Conclusion:
- If we compress the nw-builder, the execution takes forever under windows, like the Blackbox.
- If we don't compress it, the installation takes forever (it anotates all the files changed to make a clean uninstall), but the execution is almost immediate, like the Configurator.

@mikeller, can this be a secondary effect of `yarn`?

Docteh, 2019-05-08:
I don't know anything about yarn, but for some reason it looks like its just ignoring the flag to only grab production dependencies when sticking stuff into dist/node_modules

Workaround of sorts: 
Go into dists, remove ` "gulp": "^4.0.1",` from package.json, 
`rm -rf node_modules` then finally 
`yarn --ignore-scripts --production` 

9958 tiny files vs 146 tiny files

If you could get it to respect the production flag, and left the gulp dependency, you'd be down to 2872 files

Maybe its not respecting production flag only on certain situations?

Docteh, 2019-05-08:
I'll give this a go again tomorrow, with a cleanly clean repository. Hopefully its not something that only works for me...
```
[/Web/config-test] # git diff
diff --git a/gulpfile.js b/gulpfile.js
index 16825763..ecf39499 100644
--- a/gulpfile.js
+++ b/gulpfile.js
@@ -18,6 +18,7 @@ const commandExistsSync = require('command-exists').sync;
 const targz = require('targz');

 const gulp = require('gulp');
+const yarn = require('gulp-yarn');
 const concat = require('gulp-concat');
 const install = require("gulp-install");
 const rename = require('gulp-rename');
@@ -227,13 +228,7 @@ function dist_src() {
         .pipe(gulp.src('package.json', { passthrougth: true }))
         .pipe(gulp.src('changelog.html', { passthrougth: true }))
         .pipe(gulp.dest(DIST_DIR))
-        .pipe(install({
-            commands: {
-                'package.json': 'yarn'
-            },
-            production: true,
-            ignoreScripts: true
-        }));
+        .pipe(yarn({ production: true}));
 };

 function dist_locale() {
diff --git a/package.json b/package.json
index c770cb22..9a3a822e 100644
--- a/package.json
+++ b/package.json
@@ -37,7 +37,6 @@
   "author": "The Betaflight open source project.",
   "license": "GPL-3.0",
   "dependencies": {
-    "gulp": "^4.0.1",
     "i18next": "^14.1.1",
     "i18next-xhr-backend": "^2.0.0",
     "jquery-textcomplete": "^1.8.5",
@@ -48,6 +47,8 @@
     "universal-ga": "^1.2.0"
   },
   "devDependencies": {
+    "gulp": "^4.0.1",
+    "gulp-yarn": "^2.0.0",
     "chai": "^4.2.0",
     "command-exists": "^1.2.8",
     "del": "^4.1.1",
```

serran0, 2019-05-08:
> @serran0: Can you please supply some more information to help narrow this down, like what version you are trying to build, and if there is an older version that built without this issue?

I have a task that is pulling down the latest dist from GIT and builds it once a day overwriting the release installer from the day before.
Sometime between 2019-05-01 and 2019-05-06 the installer file size jumped from ~78Mb to ~90Mb and the issue with node_module folder started to occur at installation. Sorry i dont have any more info than that.

McGiverGim, 2019-05-08:
@Docteh tried your changes with the Blackbox and it seems to fix the problem too. I don't know if it is the correct way or not (I don't have `yarn` knowledge neither), but it seems to work. The `node_modules` contains only what it must.
The only problem is that I needed to start the debug with `yarn npm debug` and before that the `npm debug` worked for me. Not a problem to me.
@mikeller what do you think about the patch?

PR #1425 (2019-05-09): Fixed yarn production install.
Fixes #1423.
closed 2019-05-09 by mikeller