betaflight/betaflight-configurator #1427: Can't flash Fortini F4 OSD Ver 3 on MAC

gee4z, 2019-05-09:
Been trying to flash 4.02 onto a Fortini F4 OSD Ver3 on my Mac. last version was 3.5.5 but I updated the configurator to the latest and now can't flash even using the boot button - which, btw, I have never had to use. Mackbook Pro 2017. just keeps saying that the flash fails to connect with the board- also when it shows it's in DFU- it does nothing at all when I click the flash firmware button????

mikeller, 2019-05-09:
@gee4z: At least stating what version of the configurator you are using would go a long way towards others being able to help you.

gee4z, 2019-05-10:
Sorry @mikeller it’s 10.5.1

closed 2019-05-13 by gee4z

gee4z, 2019-05-13:
Solved it... or... just followed instructions more closely-which i don't often do... and with all previous versions all I ever had to do was plug the board into my mac and connect -no boot button or screwing with anything- now in this version I have to press boot button, plug in, close configurator and then reopen (with board still plugged in) and then I can flash new firmware. Weird how the newer configurator got more complicated...

beeb, 2019-05-23:
I can reproduce with Fortini OSD rev. 2 and Configurator 10.5.1 on macOS. Board does not actually go into DFU through the command or the "Flash firmware" button (although it says it does in the ports dropdown). The use of the boot button is required.