betaflight/betaflight-configurator #1442: No attitude feedback after Windows update

kevinjones4642, 2019-05-17:
My computer just updated Windows 7 pro, and now I can't see attitude changes in BF configurator. Has anyone else experienced this?

McGiverGim, 2019-05-17:
Try to remove the graphs and add them again. I think remember other user with the same problem fixed it in this way.

kevinjones4642, 2019-05-17:
This gets more interesting. After the Windows update, I tried 2 different quads, and in the main tab in BF configurator, the animated quad will not move, although info in all the other tabs seems to work. On the RC tab, the RC sliders move with stick/switch inputs, and on the rates tab, the animated quad moves with stick inputs. To make things more interesting, I just plugged someone else's quad into the laptop, and BF Configurator works perfectly there. I just plugged it into a bare EvoF4 board sitting on the table, and it works fine there as well.

It seems that my initial guess that a Windows update broke it is wrong.  It's just odd that it worked perfectly in a quad last night - there was just a Windows update, and now it won't work in that same quad, with nothing changed in the quad.

If anyone has ever experienced anything similar, any help figuring this out would be most appreciated. It's Configurator 10.5.1. One of the quads is pretty old, running 3.2.5, and the other is running 3.5.6. The newer one was just configured last night, and I used the animated image of the quad to fix my board orientation in software. The EvoF4 board is even older, running 3.1.7, but it's the only one that works.

kevinjones4642, 2019-05-18:
I'm not sure what you mean by "removing the graphs and adding them again."
How do you do this? I can't find any way in the configurator to add or
remove objects.

--Kevin

kevinjones4642, 2019-05-18:
OK - I officially feel stupid. On the quads I had trouble with, I had
disabled accelerometers, and this is the problem. The spare board and the
other person's quad had accelerometers turned on. Doooh! It's always the
stupid things.

Thanks for your help.

--Kevin

closed 2019-05-18 by kevinjones4642