betaflight/betaflight-configurator #1445: i need to have betaflight 10.4.0 and10.5.1 installed on my computer. currently instillation of one version removes the other version.

fLYINGfRANK1701, 2019-05-18:
i need to have betaflight 10.4.0 and10.5.1 installed on my computer at same time. currently instillation of one version removes the other version. the F3 chip will only run o n version10.4.0 of betaflight while F4 chip is compatible with 10.5.1.

McGiverGim, 2019-05-18:
10.5.1 is compatible with F3. But you can copy the installed folder to another location and then install the other version. Both will work.

mikeller, 2019-05-18:
@fLYINGfRANK1701: Configurator should be backwards compatible. If you describe what the problem is with your F3, and what hardware / firmware it is we can probably fix it and make it work with the current configurator.

fLYINGfRANK1701, 2019-05-18:
I put each version in it's own folder. when I installed one version of betaflight the other version would be deleted.  I have a modula 7, betaflight 10.4.0 works,  betaflight 10.5.1 locks up when I try to access the mode page. ill try the install again and ensure the folders are in completely separate directories. the instructions for mobula 7 say it is only compatible with betaflight 10.4.0. due to limited memory size. what a pain, flys great not sure about the setup part

McGiverGim, 2019-05-18:
I don't say that you install each one in different folders. Install one, COPY the folder with the files to other place and the install the other. You will have the two.
Your mobula does not work because it uses an unofficial/nightly version. If you update the firmware to an official version it will work with the newest Configurator.

mikeller, 2019-05-18:
@fLYINGfRANK1701: 

> the instructions for mobula 7 say it is only compatible with betaflight 10.4.0. due to limited memory size. what a pain, flys great not sure about the setup part

Looks like this is just a case of Mobula being full of sh*t and throwing their customers under the bus by shipping their products with unlicensed and broken firmware.
Install Betaflight (and not the crap that Mobula claims is Betaflight) on your hardware, and it will work just fine with all versions of configurator.

Duplicate of #1335.

closed 2019-05-18 by mikeller