betaflight/betaflight-configurator #1446: SD card: Startup failed on MATEKF722SE

smaugfm, 2019-05-19:
The card is SanDisk Ultra 16GB, I am successfully using it on another board Matek F722-STD, so the card itself is fine. 
I tried clean flashing 4.0.2, 4.0.0, the earliest one available 3.5.3, 4.1.0 #1647 from development, and also tried 4.0.0 from matek [website](http://www.mateksys.com/Downloads/FW/betaflight_4.0.0_MATEKF722SE.hex) - same results.
Tried turning sdcard_dma = ON - no luck, and tested with Lipo connected - same results.

McGiverGim, 2019-05-19:
I have this board and it works perfectly, at least with others micro sd. Have you tested others?

McGiverGim, 2019-05-19:
One thing more: I suppose your problem is with the firmware, not the Configurator, so maybe this issue is in the wrong place.

smaugfm, 2019-05-19:
I haven't actually test others, will try that. And of course this issue is not for the configurator:) I am sorry. Closing it.

closed 2019-05-19 by smaugfm