betaflight/betaflight-configurator #1447: Feature OSD Profile Names  (Planned for Betaflight 4.1)

pkruger, 2019-05-19:
Please add support for OSD Profile Names.
See:
https://github.com/betaflight/betaflight/issues/8107 and 
https://github.com/betaflight/betaflight/pull/8272

This feature adds a new OSD element (OSD Profile Name) and the ability to name each of the 3 OSD Profiles.

McGiverGim, 2019-05-20:
To have this in the Configurator, first, it needs to be in the MSP protocol in the firmware part. Is the only way the firmware and the Configurator communicate.
Usually this can be done one version later than introduced (so in theory for 4.2).

mikeller, 2019-05-20:
@McGiverGim: What you are referring to is setting the profile names in the GUI - and your statement is correct.
What can and should be done now is adding the OSD elements for these names as OSD elements to the OSD tab.

pkruger, 2019-05-21:
PR: https://github.com/betaflight/betaflight-configurator/pull/1453

closed 2019-05-21 by mikeller