betaflight/betaflight-configurator #1449: BFC 10.6 + SPRACING F3 - starts flashing but timeout at beginning of verify phase

RipperDrone, 2019-05-19:
BFC 10.6 nok, BFC 10.0 ok. Seems to do with MSP / protocol changes? BF FW version does not matter. CLI BL does not put FC into bootloader anyway, so I had to short BOOT Pins to get it flashing. Std flashing didn't wrk either then, so it needed 'flash on connect' to start the process...

mikeller, 2019-05-19:
> Seems to do with MSP / protocol changes?

Not really. But different versions of NW.js, with changes in the serial stack. Telling us what operating system you are running on would help.

> CLI BL does not put FC into bootloader

How can you tell?

One thing about flashing boards with a CP210x is that they often need 'Manual Baud Rate' to be set, 256000 seems to be working in most cases. Have you tried reducing that?

McGiverGim, 2019-05-20:
@RipperDrone  are you able to build a debug version of the Configurator yourself? Sometimes in the developers console appears the error that is happening.

closed 2019-06-26 by stale[bot]