betaflight/betaflight-configurator #1462: OSD hide logo message stays active

Jetrell, 2019-05-25:
When ever the `hide logo` tab is checked on the OSD setup page. A message will appear to instruct on this tabs use.
The problem is, that the message does not go away once the `hide logo` tab is checked. It just stays there on any of the pages in the configurator.
The message usually stays in the same location that it appears on the OSD page. But occasionally will move to the top left hand corner of the configurator.

[image]

mikeller, 2019-05-25:
@Jetrell: It would help if you at least specified what operating system / version you are using. I suspect that this is something Chrome / NW.js related, but I cannot get it to reproduce in Ubuntu, the tooltip disappears as soon as i move the mouse pointer from the 'Logo' switch over to the tab list.

Jetrell, 2019-05-25:
@mikeller It does it on Windows 7 32 and 64bit systems. With both Chrome and Windows applications.

McGiverGim, 2019-05-27:
It happened to me. I disabled the logo pressing the switch, and changed the tab and the tooltip remains. I will look into it further, but the tooltip is a external library without almost configuration.

closed 2019-05-27 by Jetrell