betaflight/betaflight-configurator #1476: Gyro Notch Filter cutoff below 100Hz reverts to Center Frequency -1 when saving

alaskatraz, 2019-06-05:
BF 3.5.7 - if I set a Gyro Notch Filter and the cutoff is less than 100Hz, it will save once, but thereafter, if I save ANYTHING, the Cutoff gets reset to the Center Frequency -1.  e.g if i set Center Freq of 150Hz and Cutoff of 90Hz, the Cutoff will revert to 149Hz if I save anything, anywhere in BF.  
I tried setting it in the CLI but the result is the same.  This only seems to happen if the Cutoff is set to less than 100Hz.
[image]

McGiverGim, 2019-06-05:
I'm not too sure that this is a Configurator bug, he says in CLI the problem is the same.

I'm trying to reproduce the bug without luck. I enter Notch 1, center 150, cutoff 90. Save. All ok. I navigate for other tabs, save several things, return to filters, and all ok. Tested with 3.5.7 and with latest master.

The cutoff is changed to center - 1 when the value of the cutoff is invalid (for example bigger than the center). 

@alaskatraz Did you flash with erase enabled? Are you using the latest Configurator? If not, please test to flash again with erase enabled to be sure that is not a corruption or similar and install the latest Configurator.

alaskatraz, 2019-06-05:
I am using configurator 10.5.1.  I only recently started tuning with Blackbox logs, so, this is my only experience settings notch filters, but it does occur on my other quads, all Tempest F4 flight controllers.  What happens is the cutoff value changes in the GUI but it is not changed in the CLI unless/until a Save operation is made in the GUI, then it saves the new cutoff value.  I did recently update them from 3.5.6 to 3.5.7 and did full chip erase.  It seems to be something in the GUI - that form field (Notch1 Cutoff) has  trouble with values less than 100 - some calculation error. It displays the erroneous value, even though it is correct in the code, but it will change the value in the code when a save event occurs. (similar to how the RTH setting would revert in Failsafe, before there was a GUI option for it).  I'll keep testing.

McGiverGim, 2019-06-05:
I need to reproduce it to be able to fix it. Maybe I did something wrong. Can you post a step by step what you do?

alaskatraz, 2019-06-05:
I just tested another board CLRacing F4.  It does the same thing.  Also, i tested Notch2 and D Term Notch and they all do the same thing.  Here are the steps:
Set Gyro Notch Filter 1 to 150 (this value is arbitrary, so far)
Set Gyro Notch Filter 1 Cutoff to 90 (or anything less than 100Hz)
Save
Click on another menu option, like "Power & Battery"
Go back to PID Tuning > Filter Settings
Cutoff Value is now 149

I repeated this for notch 2 and d term notch and the result is the same.   It seems to treat value of less than 100 the same as if it was greater than the Center frequency setting.  

Not the end of the world,  i just need to remember this in cases where i have set a notch filter this way.

alaskatraz, 2019-06-05:
Update:  If i set the center frequency to 950 and then set the cutoff to 90 it does not change.  So, there is something that is reading 90hz as 900Hz in the cutoff field, i think.

McGiverGim, 2019-06-05:
I think it is comparing values in ascii and not in numeric form. I'm out of the computer, I will try to reproduce it tomorrow.

alaskatraz, 2019-06-05:
OK. Thank you.

Here is a video of it.
https://youtu.be/YHqT_lVfYYc

Rick Dennis
The I.T. Department
907-243-8001

McGiverGim, 2019-06-06:
Finally I have reproduced it. It only happens with version 10.5.1, with code in latest master it does not happen (it was fixed at some point).

@mikeller I don't know if it is enough important to release a hotfix version or better wait until next official release.

mikeller, 2019-06-06:
Not sure if this merits getting all users to update configurator just for the few who are using a deprecated firmware version.

Also, if we want to release a patch version, we'll first have to identify which change(s) are involved so that we can pull them into the maintenance branch.

alaskatraz, 2019-06-06:
It doesn’t matter to me.  I just wanted to report the bug.  If it has been addressed/resolved, that is fine.

Is there something I can do to get a version that doesn’t have this issue?  I’m not clear on that part.

Thank you.

Rick Dennis
907-243-8001

McGiverGim, 2019-06-06:
If you use Windows, you need to compile yourself from master. If not you can get it from here https://ci.betaflight.tech/

alaskatraz, 2019-06-06:
OK. Thank you.

Rick Dennis
907-243-8001

McGiverGim, 2019-06-06:
Just a video: https://youtu.be/OpE7E89ftHY

alaskatraz, 2019-06-08:
Thanks!  That did the trick – all good now.

Rick Dennis

closed 2019-06-08 by McGiverGim