betaflight/betaflight-configurator #1481: Autocomplete refinements

bizmar, 2019-06-07:
Two things that annoy me in everyday usage:

1. Leaving the mouse above the input field, highlights an item on the popup menu, that appears underneath it. if you move the mouse away the last option stays active. [Example](https://v.usetapes.com/C3DZ771I6Q). 
Proposed solution: Only select option under mouse if mouse has moved. Or adapt how Intellisense works, and only highlight on a mouse click, and select with a double click. [Example](https://v.usetapes.com/QzkzxutfPK)

2. Entering commands while autocomplete is being build. Even though there is already some mechanisms to discourage writing before it is completed, I still manage to do it somehow. 
Proposed solution: Disable "enter" while the list is being made. Or cache the commands so the list does not have to be built every time. 

Tagging @Cleric-K

Cleric-K, 2019-06-07:
1. I'll have to check about that one. It is behavior coded in the library that we use. The question is whether we can modify the behavior without having to patch the library itself.

2. Can you make a detailed example for this? (like the screencast for 1).
Currently the input is _disabled_ (it should look grayed out) while the list is being built. I don't see how it is possible to enter any commands.

bizmar, 2019-06-07:
2. I figured out what's happening...  first part of the command is lost due to the field being disabled, and remaining part is invalid. So this is user impatience. 😅 Probably the right way to fix it, is to "cache" what is written and only "let it through" once building is done. No idea how feasible this is.

I love the feature, but it's also a bit in my face all the time. If I "invoke it", and then delete what I wrote, it will still try to autocomplete the first command, and I have to `esc` to get rid of it. But I can live with all of that.

The 20% polish that can take 80% of the time dev to get right. 😉 I would really appreciate fixing the mouse thing, what usually happens is that I'm using `get` as a search, press enter and the wrong thing comes out because of where left the mouse.

I should really open a request to expose some of what I search frequently in the UI.

mikeller, 2019-07-30:
@bizmar: Has this been addressed in the updates to autocomplete?

bizmar, 2019-07-30:
i'm not aware that any of these have been adressed.

Cleric-K, 2019-07-31:
No, sorry, I haven't had the time to experiment with this yet.

Cleric-K, 2019-08-01:
About problem "2" I think it is not worth the effort. Trying to cache keystrokes while the input being disabled could turn out to be quite an adventure :D

bizmar, 2019-08-01:
Using it now again for a few days. The mouse position issue gets me a lot. I'm fine with the second one. Thanks for looking into it.

I use Debug modes a lot. Having them not be in the same order is slightly confusing. But what would really make this feature awsome, would be, if it would weigh the results based on frequency of use. So the long debug list would have my most used modes up top. I appreciate that this is also not trivial. And I would rather other parts of the configurator be made as slick and usable as the CLI 😜

Cleric-K, 2019-08-01:
@bizmar I guess you have seen #1552 that addresses the mouse thing. You can give it a test.

About the debug modes - sorting by use will add a ton of complexity and I don't really see the point. 

Wanting the modes to be on top makes sense only if you want to reach them more quickly with `down` arrow. But writing just 2-3 letters of _any_ part of the mode name will usually filter the list up to the single mode you are looking for. So why not use the letters? Or I'm not getting your question?

bizmar, 2019-08-01:
The thing is, that the debug names are often confusing and hard to remember for me, I actually kinda remember them by Location in the get command readout. I know this is specific to me, so feel free to ignore. Thanks a lot for fixing the mouse position edge-case.

closed 2019-08-01 by bizmar