betaflight/betaflight-configurator #1482: Firmware fashing still fails on Mac OS X

GB509, 2019-06-07:
Using  the betaflight configurator 10.5.1 (a769d38) on Mac OS 10.13.6 still fails to flash firmware upgrade in my case of the CLRACING F4

erakli, 2019-06-10:
As a workaround, take a look at [`stm32flash`](https://sourceforge.net/p/stm32flash/wiki/Home/).

etracer65, 2019-06-12:
There are no problems with the Configurator. I flash dozens of times a day using the 10.5.1 configurator on MacOS 10.13.6.

You need to better describe "fails to flash" and give a detailed description of what happens and any error messages.

If the flight controller fails to enter DFU mode and just appears to disappear from the ports when starting to flash, then that's a problem with the devices you have connected to the flight controller interfering with the boot loader. Disconnect or make sure power is removed from any receivers, GPS modules, or anything else connected to a UART.

MichaelManz, 2019-06-23:
@etracer65 thanks! Great advice, helped me with my DFU problems

etracer65, 2019-06-23:
Closing due to lack of response for information requests.

closed 2019-06-23 by etracer65