betaflight/betaflight-configurator #1489: Bottom bar has moved down

Docteh, 2019-06-13:
[image]

Some sort of CSS issue.  `div with class of content_toolbar` is being tucked underneath the `status-bar` instead of above.

My suggestion would be to add more padding to the bottom, like `padding: 10px 0 24px 0;` But that doesn't fix whatever caused this, and doesn't fix the cli tab.

https://github.com/betaflight/betaflight-configurator/blob/bb72a44f84ec0b403647f74554d01087c2c88104/src/css/main.css#L1042-L1051

McGiverGim, 2019-06-13:
I'm trying to reproduce the problem without luck. Using the minimum size that the Configurator lets, my buttons are in place:
[image]

Can you explain step by step how to get to this issue?

Docteh, 2019-06-13:
Thanks. I'll look into it and let you know. It could be

~~1) extra files~~
~~2) I'm using the x64 development version of NWJS 0.38.2~~
3) A [Byte order mark](https://en.wikipedia.org/wiki/Byte_order_mark) in main.html that gets stripped somewhere in the gulp process. But wont get stripped if someone were use rsync or robocopy. 

```
[/Web/betaflight-configurator] # hexdump -C src/main.html
00000000  ef bb bf ef bb bf 3c 21  44 4f 43 54 59 50 45 20  |......<!DOCTYPE |
00000010  68 74 6d 6c 3e 0a 3c 68  74 6d 6c 3e 0a 3c 68 65  |html>.<html>.<he|
```

PR #1493 (2019-06-15): remove extra Byte order mark from beginning of src/main.html
If the extra byte order mark in src/main.html makes it to chrome, there is a rendering issue. 
Fixes #1489 

Before:
```
[/Web/betaflight-configurator] # hexdump -C src/main.html  | head
00000000  ef bb bf ef bb bf 3c 21  44 4f 43 54 59 50 45 20  |......<!DOCTYPE |
00000010  68 74 6d 6c 3e 0a 3c 68  74 6d 6c 3e 0a 3c 68 65  |html>.<html>.<he|
00000020  61 64 3e 0a 20 20 20 20  3c 6d 65 74 61 20 68 74  |ad>.    <meta ht|
00000030  74 70 2d 65 71 75 69 76  3d 22 43 6f 6e 74 65 6e  |tp-equiv="Conten|
00000040  74 2d 54 79 70 65 22 20  6 [... cut]
closed 2019-06-15 by mikeller