betaflight/betaflight-configurator #1490: OSD element "RSSI DBM Value" needs to be added to configurator

Docteh, 2019-06-13:
https://github.com/betaflight/betaflight/pull/8309

That commit adds an element, an alarm, and a warning.

McGiverGim, 2019-06-13:
Yes, I saw it yesterday. Any OSD element that has not been added to the Configurator appears as unknown. Is a way to not forgot any before the next release :)

I didn't add it yesterday because I prefer to merge first this: https://github.com/betaflight/betaflight-configurator/pull/1487 if not it will conflict.

But is good to have it here as an issue to not forgot about it 😄

mikeller, 2019-06-13:
#1487 has been merged.

McGiverGim, 2019-06-13:
I created the PR with the preview and the warning. I can't add the alarm because it needs to be added to the MSP first.

closed 2019-06-15 by mikeller