betaflight/betaflight-configurator #1495: [FR] Add functionality to get motor noise value

bosbefok, 2019-06-13:
In the motors tab of the configurator, add functionality to test an individual motor and receive a calculated noise value/score. 

This would be useful as it would allow a user to test new motors and receive a baseline value, and then monitor how the value degrades over time after crashes and normal usage.

mikeller, 2019-06-15:
@bosbefok: Can you describe a bit more what data you would be using to calculate the noise value, and how you would collect this data?

bosbefok, 2019-06-15:
Based on my limited knowledge, I'd assume it would be capturing gyro data, as well as variations to the axes. I understand this data can already be seen in the profile graph on this screen, but my suggestion is more of a standard and repeatable test - for example a 2 second test running at 50% power, 75% power, 100% power and then a score generated by some sort of average of the values. 

This is different to spinning each motor up individually (as most users currently do) as it would ensure the same repeatable conditions are used for each test, and gives more of a basis for comparison.

Perhaps you could elaborate whether you think this is feasible.

haslinghuis, 2024-06-23:
Closing as telemetry is shown in motors tab for some time now.

closed 2024-06-23 by haslinghuis