betaflight/betaflight-configurator #1498: Unable to use on Win 7

dannydel, 2019-06-16:
Hello,

I'm writing this for my father who is currently trying to setup betaflight onto his windows machine. He installed 10.5.1 and it was working well. But he moved some files around by mistake and it caused a weird issue where it would load up but no items and no tabs would load. All the buttons do not work either. We uninstalled it and reinstalled it to no avail. Still dead pages appeared. 

Is there any steps we can take so that he can get betaflight to work on his machine because it seems like it is keeping files even after we uninstall and delete the folder in the Program Files 86x folder. 

We used the system on my Mac and its a pretty sweet set of software!

Cheers,
Danny

McGiverGim, 2019-06-16:
Uninstall it, delete the remaining files (or the entire folder) manually and install it again. It uses only the files from the install folder, so it is doesn't work the problem is in another place.

Docteh, 2019-06-28:
Maybe clear out
`%LOCALAPPDATA%\betaflight-configurator\User Data`

Not sure about variables and Windows 7, might need to manually look for like
`C:\Users\Docteh\AppData\Local\betaflight-configurator\User Data`

closed 2019-08-04 by stale[bot]