betaflight/betaflight-configurator #1500: [Windows] Stuck at connect for FC with empty signature/manufacturer_id

DieHertz, 2019-06-17:
I have an FC here which multiple users (including myself can't connect to on Windows. Problem doesn't occur on macOS, and it seems to be related to these FCs having empty strings as both signature and manufacturer id.
When configuring the same FC as a unified target everything works on both Windows and macOS.

I localized it to being related to board info since configurator gets stuck at `MSP_BOARD_INFO` most of the time.
To work around this, I had to undefine `USE_SIGNATURE` and `USE_BOARD_INFO` for testing purposes, maybe someone knows a better way?

mikeller, 2019-06-17:
@DieHertz: Hey, welcome back, long time no see. :-)

What target is this happening with?

Also, do you happen to be able to capture a console log of when this is happening (requires the debug version of configurator)?

DieHertz, 2019-06-18:
It's a not yet released AIKONF7, defined in a branch here:
https://github.com/DieHertz/betaflight/tree/aikon-f7-target
https://github.com/DieHertz/betaflight/commit/37b8c0bc1cde0e1382c90d84f2ce6daad408b475

I've double and triple-checked the target files and couldn't spot the mistake, but maybe I'm missing something.

Will capture the log when I'm home.

mikeller, 2019-06-18:
Interesting one. I have flashed your target (which looks completely ok) to a stand-in board, and this connects just fine to configurator in linux and Windows - so there seems to be at least some hardware dependency at play.

closed 2019-07-25 by stale[bot]