betaflight/betaflight-configurator #1502: [FR] Add Blackbox debug modes to the Blackbox tab.

bizmar, 2019-06-23:
Add Blackbox debug modes to the Blackbox tab. Probably as a dropdown menu. Maybe incorporate or link to this wiki article. And maybe order the list the same way as it's in the article, but if that prevents it from being auto generated, then go for the least amount of future maintainability. :)

Maybe make Gyro_scaled the default, instead of none? (putting this here to start a discussion, please don't disregard the while request because of this part)

I'm requesting this addition because if I ask someone to make me a BB log, they have to go into 3 different tabs to set it up in a meaningful way.

mikeller, 2019-06-23:
This will first have to be added to MSP in the firmware.

etracer65, 2019-06-24:
We don't want to set a default `debug_mode`. Doing so will increase the log size and cause a performance hit (particularly for things related to the gyro).

bizmar, 2019-06-24:
> We don't want to set a default `debug_mode`. Doing so will increase the log size and cause a performance hit (particularly for things related to the gyro).

Yes, it uses more resources but it also gives valuable data, and if logging with it enabled, actually affects flight performance... well, that raises big questions...

Having debug modes in the same tab will be enough for now. 😄

PR #1566 (2019-08-15): Add debug modes to blackbox tab
Fixes #1502
Adds debug mode setting in the blackbox tab. Requires MSP to go with it which will be added.
closed 2019-08-15 by mikeller