betaflight/betaflight-configurator #1503: motor direction switch in latest win GUI missing !!!

yo8aiv, 2019-06-23:
as title says motor direction switch in GUI is missing
the interface is supposed to make life easyer right.

same thing i have seen missing in latest inav win GUI interface app aswell.
its annoying because when i fly low all the grass get's thrown as a salad in the camera lens so i prefer reversing motor rotation to avoid it.
https://user-images.githubusercontent.com/24196117/59229319-52375480-8bda-11e9-85bc-51e306f618ef.jpg

meaning i prefer having a graphical representation of how motor direction is set in the FC.
if this has been submitted just link the issue.
thanks.

McGiverGim, 2019-06-23:
This is Betaflight, not iNav. Maybe wrong place to put the issue?

yo8aiv, 2019-06-23:
sorry corrected .. sort of my way of expressing the issue (not my main language eng!!). But no its related to BF since the image is BF gui.

mikeller, 2019-06-23:
@yo8aiv: As your screen shot shows, the option is clearly there in Betaflight configurator. I suspect that you are trying to use it with firmware that is too old, and does not support setting this option through the GUI. You have to update the firmware, or use the CLI to set this option `yaw_motors_reversed`.

etracer65, 2019-06-24:
Maybe you're confused by the function of this switch? It just tells the firmware that you have reversed the direction of the motors. It does not cause the motors to be reversed.

yo8aiv, 2019-06-27:
hello. thanks for your answers.
 yes i admit i got a bit confused with all the versions ... win and chrome.  
the fw wasn't too old was 3.5.3 wich had this enabled at the time i had everything configured. 
i will properly test everything and report back.

yo8aiv, 2019-07-22:
Just a suggestion.  I knew that .. but may of my friends didn't . so can a HINT be added to that switch with this/your  phrase : 

>  It just tells the firmware that you have reversed the direction of the motors. It does not cause the motors to be reversed.

closed 2019-07-22 by yo8aiv