betaflight/betaflight-configurator #1519: Abortion while loading hex file

TheIsotopes, 2019-07-02:
I always have problems loading hex files.
Loading is interrupted very often, only after repeated loading the hex file is released for flashing.

In the console, I then always find this error message:
`firmware_flasher.js:385 File limit (1 MB) exceeded, aborting`

https://github.com/betaflight/betaflight-configurator/blob/c167f48c7311d77e0bbb1a675187171e3f4537fb/src/js/tabs/firmware_flasher.js#L382-L388

Maybe you should increase that limit now.

McGiverGim, 2019-07-02:
Ah! Interesting. A lot of F7 files are greater than 1Mb now... what I don't understand is that trying again it works :)

TheIsotopes, 2019-07-02:
I did not understand either ... loading the hex file several times or restarting the configurator solve the problem

TheIsotopes, 2019-07-02:
Here you can see an complete console output ... you can see three tries then a reconnect and also three tries until it works

```
i18next::backendConnector: loaded namespace messages for language en Object
i18next.min.js:1 i18next: languageChanged en
i18next.min.js:1 i18next: initialized Object
localization.js:32 i18n system loaded
main.js:145 Libraries: jQuery - 2.1.4, d3 - 3.5.6, three.js - 72
port_handler.js:74 Selecting last used port: COM6
analytics_debug.js:16 
                         _                          _       _   _
                        | |                        | |     | | (_)
  __ _  ___   ___   __ _| | ___    __ _ _ __   __ _| |_   _| |_ _  ___ ___
 / _` |/ _ \ / _ \ / _` | |/ _ \  / _` | '_ \ / _` | | | | | __| |/ __/ __|
| (_| | (_) | (_) | (_| | |  __/ | (_| | | | | (_| | | |_| | |_| | (__\__ \
 \__, |\___/ \___/ \__, |_|\___|  \__,_|_| |_|\__,_|_|\__, |\__|_|\___|___/
  __/ |             __/ |                              __/ |
 |___/             |___/                              |___/

analytics_debug.js:16 Running analytics_debug.js. This script is intended for testing and debugging only.
log @ analytics_debug.js:16
analytics_debug.js:16 Initializing Google Analytics.
analytics_debug.js:16 Tracking script already loaded. Abandoning initialization.
log @ analytics_debug.js:16
analytics_debug.js:16 Registered new plugin: ga(provide, "render", Function)
serial_backend.js:59 Connecting to: COM6
serial.js:138 SERIAL: Connection opened with ID: 1, Baud: 115200
MSPHelper.js:1351 Arming disable
MSPHelper.js:1354 Real time clock set
three.min.js:536 THREE.WebGLRenderer 72
MSPHelper.js:1351 Arming disable
serial.js:266 SERIAL: Connection with ID: 1 closed, Sent: 1144 bytes, Received: 9528 bytes
FirmwareCache.js:231 Firmware cache journal loaded; number of entries: 0
firmware_flasher.js:376 Loading file from: ~\Modellbau\QuadCopter\betaflight_4.1.0_MATEKF722SE-1729-5da85ef4a.hex
firmware_flasher.js:385 File limit (1 MB) exceeded, aborting
firmware_flasher.js:376 Loading file from: ~\Modellbau\QuadCopter\betaflight_4.1.0_MATEKF722SE-1729-5da85ef4a.hex
firmware_flasher.js:385 File limit (1 MB) exceeded, aborting
FirmwareCache.js:231 Firmware cache journal loaded; number of entries: 0
firmware_flasher.js:376 Loading file from: ~\Modellbau\QuadCopter\betaflight_4.1.0_MATEKF722SE-1729-5da85ef4a.hex
firmware_flasher.js:385 File limit (1 MB) exceeded, aborting
FirmwareCache.js:231 Firmware cache journal loaded; number of entries: 0
serial_backend.js:59 Connecting to: COM6
serial.js:138 SERIAL: Connection opened with ID: 2, Baud: 115200
MSPHelper.js:1351 Arming disable
MSPHelper.js:1354 Real time clock set
three.min.js:536 THREE.WebGLRenderer 72
MSPHelper.js:1351 Arming disable
serial.js:266 SERIAL: Connection with ID: 2 closed, Sent: 874 bytes, Received: 6731 bytes
FirmwareCache.js:231 Firmware cache journal loaded; number of entries: 0
firmware_flasher.js:376 Loading file from: ~\Modellbau\QuadCopter\betaflight_4.1.0_MATEKF722SE-1729-5da85ef4a.hex
firmware_flasher.js:385 File limit (1 MB) exceeded, aborting
firmware_flasher.js:376 Loading file from: ~\Modellbau\QuadCopter\betaflight_4.1.0_MATEKF722SE-1729-5da85ef4a.hex
firmware_flasher.js:385 File limit (1 MB) exceeded, aborting
FirmwareCache.js:231 Firmware cache journal loaded; number of entries: 0
firmware_flasher.js:376 Loading file from: ~\Modellbau\QuadCopter\betaflight_4.1.0_MATEKF722SE-1729-5da85ef4a.hex
firmware_flasher.js:385 File limit (1 MB) exceeded, aborting
firmware_flasher.js:392 File loaded
hex_parser.js:106 HEX_PARSER - File parsed in: 0.1850 seconds
```

McGiverGim, 2019-07-02:
Is strange, this code is only executed when loading a local file (not an internet one) and it lets to me flash it after all...

McGiverGim, 2019-07-02:
Ok, it seems that this is readed by chunks, so sometimes it reads the file completely, it detects the file is bigger than expected, and aborts too late, the file is in memory. The flash process only detects if the file has been readed completely or not to start flashing.
Maybe in your case, the chunk is different each time and sometimes it aborts and sometimes not. 
I will try to fix it, giving more size and validating the things right.

closed 2019-07-04 by mikeller