betaflight/betaflight-configurator #1524: How to flash BF on STM32F411 Discovery

moge11, 2019-07-04:
Hello,
I'm in possession of a STM32F411 Discovery board and I saw that there exists a  BF firmware for it, so I thought why not try to use it as a FC, but I'm now trying since two days to upload BF on my board without success. I always get the same error: No response from the boot loader. Programming failed. I followed the official instructions on how to flash it, which are a bit spare for this board, and tried several different methods, I updated all drivers and got the board into boot mode, but it never displays in Zdig, it only shows the programmer ST-LINKV2 even if i plug it into the user USB.

Thanks in advance

mikeller, 2019-07-04:
@moge11: You will have to flash with the ST-Link utility (https://www.st.com/en/development-tools/stsw-link004.html) at least for the first time, or you will have to put the STM32F411 into DFU mode by connecting the BOOT0 pin to VCC, and then use the Betaflight configurator.

moge11, 2019-07-05:
@mikeller: and where do i get the .hex or binary file,whuch is needed for flashing?

jflyper, 2019-07-05:
Goto Betaflight github repo, and select Releases under code.

closed 2019-07-12 by moge11

embhobbb, 2020-06-27:
Is this the file **betaflight_4.2.0_STM32F411.hex**  for discovery board F411?

mikeller, 2020-06-28:
@embhobbb: Yes.