betaflight/betaflight-configurator #1538: Gyro Notch Filter cutoff to 100Hz Cutoff reverts to Center Frequency -1 when saving

spatzengr, 2019-07-18:
Related to https://github.com/betaflight/betaflight-configurator/issues/1476.  If you set the Notch Center to 99 or below, all is well.  However, if exactly 100hz for the center, it switches the cutoff to 99hz.

Docteh, 2019-07-19:
Hi, I forgot to ask on slack if you had tried anything newer than the 10.5.1 configurator. 
I'm pretty sure this got fixed in the master build of the configurator back in april.

Since there is no windows build and I'm not a fan of forcing the chromeos version to run as an extension I gave the following a shot:

* https://nwjs.io/ Grab a normal build, unzip it somewhere
* https://ci.betaflight.tech/job/BetaFlight_Configurator/ Grab the chromeos.zip, unzip that into the same folder.
* Run nw.exe

mikeller, 2019-07-23:
@Docteh:

> Since there is no windows build and I'm not a fan of forcing the chromeos version to run as an extension I gave the following a shot:

Any reason for that?

Albeit Google has announced the sunsetting of Chrome web apps for non-ChromeOS a while back, and has stopped accepting new Chrome web apps for non-ChromeOS in their web store, support for them in Chrome itself is still going strong. Seeing that they managed to get some foothold with Chromebooks in schools (and taking a chunk out of a segment previously often ruled by Apple), I think they will not be going away any time soon, and in order to keep the wheels of the associated development community going I suspect they will have to keep support for running Chrome web apps locally in Chrome for the time being.

Docteh, 2019-07-24:
Two things really bothered me
* Chrome complained about having a developer extension loaded every time I opened it.

* I had to save a bookmark to chrome://apps because they took away other references to it.

Although looking again I could have saved a link to the app on my desktop. But maybe one day they'll break those :-/

On a related note, there has been some recent movement on accessing serial devices
https://bugs.chromium.org/p/chromium/issues/detail?id=884928#c44

mikeller, 2019-07-25:
@Docteh: I am using Chromium (on Linux), and it seems to have none of these two issues. :grin:

spatzengr, 2019-08-29:
This issue appears to be resolved as of commit 9d6f35c7 (Configurator 10.6.0 nightly build testing).

smitty97, 2019-09-04:
Also happening to me with center at 120hz, cutoff at 60hz.  Might not be dependedn on being exactly 100hz.  Cutoff is becoming 119 in user interface only, if i don't click 'save' there it stays at 60 when i inspect in CLI

closed 2019-10-11 by stale[bot]