betaflight/betaflight-configurator #1541: 10.5.1 on Ubuntu 18.04: Firefox goes to 100% CPU, on many process, as soon as betaflight-configurator starts

RRAlex, 2019-07-22:
This one sounds crazy and doesn't make sense to me, but is 100% reproducible here:

If I start betaflight-configurator 10.5.1 via sudo (as root), my normal user's Firefox's (68.0) process start acting up and consume a lot of CPU for like 5 minutes, on like 10+ sub-process.
This happens even if I have quit betaflight-configurator right after starting it...

I tested repeatedly and same result.
I'm guessing that on start it tries to check/trigger/configure something that uses Firefox somehow, maybe indirectly through Gnome or something related to its library / desktop environment.

If I can test something specific, please let me know!

mikeller, 2019-07-22:
Some questions:
- can you specify a bit closer what OS / version you are using;
- what version of configurator (.zip / .deb / .rpm / Chrome Web app) are you using;
- why are you trying to run configurator as root - this seems to be wrong in any case.

RRAlex, 2019-07-23:
```
> lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

> dpkg -l betaflight-configurator
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                          Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  betaflight-configurator                       10.5.1                      amd64                       Crossplatform configuration tool for Betaflight flight control system.
```
And I'm running as root to access ttyACM0 in a lazy manner (yes I could add myself to the right group) :)

mikeller, 2019-07-25:
Tried running Firefox while starting Betaflight configurator (not as root), and this is working just fine.
Running anything that does not need privileged access as root is wrong, much less something potentially as exploitable as a NW.js application, you should just fix this, and run stuff the way it is supposed to be run.
Closing this.

closed 2019-07-25 by mikeller

RRAlex, 2019-07-25:
@mikeller ok, but how come the package is not working as a user then?
What is needed that it's not doing on install?

mikeller, 2019-07-26:
@RRAlex: The package is working just fine as a user.
If the user you are using does not have the permissions required to access the device that the flight controller is connected as then the system you are using isn't properly configured for this to work, but this is not a problem of the configurator package.

Maybe you should read the instructions first, and then complain: https://github.com/betaflight/betaflight-configurator#betaflight-configurator