betaflight/betaflight-configurator #1546: feature request: P/D ratio link control for pitch/roll axises

dogjutsu, 2019-07-28:
The current consensus on PID tuning is that once you "find" a good P/D ratio you keep that ratio then raise both in proportion to deal with propwash and such (re: UAV Tech tuning videos).

It would be quite convenient if there were a GUI control for Roll and Pitch that would "link" the current P/D ratio so as you change one, the other also changes to maintain the ratio at which the ratio link was set -- perhaps even display the ratio itself as a value that you can tweak directly when enabled (though a decision would have to be made as to which value changes when this ratio number were changes -- probably P).

It would save a bunch of manual and error prone calculations at the 'tuning out propwash' phase.

corrado-c, 2020-12-07:
fortunately I found that a request equal to my idea has already been made, I can only queue.
Who like me uses only the sliders to tune the quad,
he realizes that something is missing to adjust the ratio between pitch and roll. Now the sliders disappear if you manually change a PID parameter. With a special field, or even something in the CLI to adjust this ratio can be very convenient.
Thank you so much for everything!

ctzsnooze, 2021-09-03:
This is being addressed very effectively in the new PID slider model.  It will be easy to add more/less P or D on pitch, relative to roll.  The user would tune roll first then adjust pitch as required.  After that point those proportions would be locked in during subsequent 
See:
https://github.com/betaflight/betaflight/pull/10919
https://github.com/betaflight/betaflight-configurator/pull/2572

closed 2024-05-06 by haslinghuis