betaflight/betaflight-configurator #1548: 10.5.1 on Ubuntu 18.04: Not getting left side menu nor welcome screen content

RRAlex, 2019-07-29:
As a follow up to #1541, I'm attaching what happens when I start betaflight-operator as a user that has the correct dialout group setup:

[image]

Knowing that it works as root, there has to be something else missing in term of access or other setting that makes the menu and the welcome page disappear?
What can I test? :)

cheers!

mikeller, 2019-07-29:
@RRAlex: Can you please post logs in text form instead of as screenshots? Nobody wants to type your error messages in order to google them.

Docteh, 2019-07-30:
Where exactly did you get that from? Is this the .deb or .rpm on the releases tab?

I get a screen like that when I break the javascript files, or forget to provide a version.json.

RRAlex, 2019-08-02:
> Where exactly did you get that from? Is this the .deb or .rpm on the releases tab?
> 
> I get a screen like that when I break the javascript files, or forget to provide a version.json.

It's the deb from github, nothing out of the ordinary and didn't modify anything.

Docteh, 2019-08-02:
When I do 
/opt/betaflight/betaflight-configurator/betaflight-configurator --password-store=basic

I only get an error about the component_loader, but the app does load. 

[image]

I'd try removing the following
`~/.config/betaflight/betaflight-configurator`
`~/.cache/betaflight/betaflight-configurator`

And if that still doesn't work, maybe try and `cat /opt/betaflight/betaflight-configurator/version.json` as a user.

mikeller, 2019-08-03:
@RRAlex: Googling your error message returns https://simpleit.rocks/linux/ubuntu/fixing-common-google-chrome-gpu-process-error-message-in-linux/ as the top result. I'd try using the advice given there, and run `/opt/betaflight/betaflight-configurator/betaflight-configurator --disable-gpu --disable-software-rasterizer`. It looks like your hardware does not have a GPU.

RRAlex, 2019-08-04:
> @RRAlex: Googling your error message returns https://simpleit.rocks/linux/ubuntu/fixing-common-google-chrome-gpu-process-error-message-in-linux/ as the top result. I'd try using the advice given [...]

That didn't work, but interestingly I have the same behaviour (user not working, root working) on a Mint 19 machine. There must be something beside the dialout group that is missing...

Docteh, 2019-08-04:
Does chrome itself work on these computers? 

If I had this happen, I'd copy /opt/betaflight/betaflight-configurator to somewhere in my user dir

Then grab an SDK build of NWJS https://nwjs.io/downloads/ and drop that on top of the copy of the configurator, and then try running the binary. I'm on windows today and its `nw.exe` so I'd guess it's just `nw`. Then when you're looking at that screen that doesn't load properly you can right click the window, select `Inspect`, and then take a look at the `Console` tab of the development tools.

RRAlex, 2019-08-05:
Thanks for the trick :)

So, I'm getting this on the nwjs console:
`main.html:1 Error handling response: TypeError: Cannot read property 'userLanguageSelect' of undefined at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/localization.js:122:20`

... and:
`main.html:1 Unchecked runtime.lastError: IO error: .../000007.log: Unable to create sequential file (ChromeMethodBFE: 7::NewSequentialFile::5)`

First one made me wonder if this was because I wasn't in a known default locale, I tried:
export LC_ALL=en_US.UTF-8
export LANG=en_US.UTF-8
export LANGUAGE=en_US.UTF-8
... and launched the betaflight-configurator again, but that didn't help.

Do any of these errors sound familiar?
thanks!

McGiverGim, 2019-08-05:
It seems the problem is here:
https://github.com/betaflight/betaflight-configurator/blob/0f04022b0c027930b6d662c76377b82f03c82605/src/js/localization.js#L122

But I don't know why exactly.

McGiverGim, 2019-08-05:
You can change this for

`if (result && result.userLanguageSelect)`

And it will work. Can you test it?

mikeller, 2019-08-05:
@RRAlex: This is highly odd. Can you describe what hardware (CPU, graphics adapter) it is that you are experiencing these problems on?

RRAlex, 2019-08-05:
So... I managed to fix it.
One of the file in ~/.config/betaflight-configurator/ wasn't owned by my user because of previous attempts.
But I still thing there is something to fix:

When it started I was seeing a popup go under the (then broken) loading window and it didn't stay open long enough to be readable. It wasn't there when I alt-tab or minimized the other window.
So using a desktop recorder I managed to read it:
```
Your preferences cannot be read.
Some features will be unavailable and changes to preferences won't be saved.
[ ] Send feedback to help us fix this issue.
[ OK ]
```
Which made me check the preferences and fix them...

So I guess it's a user error at this point, but had I seen this popup stay open, I would have figured it out earlier!
Is it possible there is a bug with the warning disappearing as soon as the other window opens?

Cheers! :)

mikeller, 2019-08-05:
@RRAlex: Ha, glad to hear that it's working. Getting messed up permissions on shared user-level config / cache files is one reason why it is not advisable to run pretty much anything as root in unix.

The disappearing error message that you are getting is usually caused by your preferences having been written by a newer version of NW.js, probably when you were trying out things in order to get this to work. Simply delete `~/config/betaflight-configurator/` to make it go away - there aren't many preferences that you'd want to keep in there anyway.

RRAlex, 2019-08-06:
> The disappearing error message that you are getting is usually caused by your preferences having been written by a newer version of NW.js, probably when you were trying out things in order to get [...]

Now that I know, I'll fix my other machine by chmod-ing it.
But that said, if someone else hits a corrupt/permission problem in that folder, (Which I had before the NW.js installation) they also might not be able to read the popup in time?
I'll close the issue, but this might need some testing for the warning to work as intended.
Cheers! :)

closed 2019-08-06 by RRAlex

mikeller, 2019-08-07:
@RRAlex: The warning is generated by NW.js, so you'll have to raise this with the NW.js maintainers to get it fixed.

mikeller, 2019-08-27:
@RRAlex: Please read https://github.com/betaflight/betaflight-configurator/issues/1548#issuecomment-519036808.