betaflight/betaflight-configurator #1557: [FR] Display Motor RPM in the Motors tab

bizmar, 2019-08-05:
Use ESC telemetry or Dshot RPM Feedback to display the current RPM per motor, or average off all motors. Useful for diagnostics purposes, and for determining a good idle RPM for the new Dynamic Idle feature.

McGiverGim, 2019-08-05:
Is there some MSP command that returns this value? If true, I can add it without problem.

joelucid, 2019-08-05:
I don't think there is but it could be added easily. The info is available in rpm_filter.c where it gets set as rpm_filter debug. Would only need to get inverted into rpm from rps - if that's what we really want.

mikeller, 2019-08-05:
I think this could be valuable as a verification that bidirectional Dshot is working correctly outside of CLI.

Mateyhv, 2019-09-24:
Agree, that definitely would be a very nice addition to the configurator.

McGiverGim, 2019-09-24:
This was added some weeks ago. I close this request.

closed 2019-09-24 by McGiverGim