betaflight/betaflight-configurator #1563: Failed to open serial port window10

ledevpole, 2019-08-07:
*Hi!*
# Failed to open serial port 
## Trying to update my firmware i have this error happennig: Failed to open serial port
* For normal configuration everythings fine and works. 

* Here are the logs:( sorry it's in french)
```
2019-08-07 @ 18:18:47 -- Veuillez mettre à jour le microprogramme pour accéder à la rubrique Flasher un microprogramme tab.
// ask for ubdate of the B. configurator that i already done.
//ok, i juste updated you..

// Here that ask me to update the configurator, to access the "flash your micro program tab", that i just done.
2019-08-07 @ 18:20:36 -- Veuillez mettre à jour le microprogramme pour accéder à la rubrique Bienvenue tab.
2019-08-07 @ 18:20:43 -- Armé
2019-08-07 @ 18:20:43 -- Takeoff Prevention activé
2019-08-07 @ 18:20:43 -- Port série fermé avec succès.
2019-08-07 @ 18:20:51 -- Utilise les informations en cache pour jobs.
2019-08-07 @ 18:20:52 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:20:52 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:21:31 -- Utilise les informations en cache pour jobs.
2019-08-07 @ 18:21:32 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:21:32 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:21:35 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:21:37 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:21:52 -- Utilisation des données de la release en cache pour firmware release.
2019-08-07 @ 18:21:55 -- Echec d'ouverture du port série
2019-08-07 @ 18:22:16 -- Echec d'ouverture du port série
2019-08-07 @ 18:22:19 -- Echec d'ouverture du port série
2019-08-07 @ 18:22:22 -- Echec d'ouverture du port série
2019-08-07 @ 18:22:23 -- Echec d'ouverture du port série
2019-08-07 @ 18:22:25 -- Echec d'ouverture du port série
```
[image]

### If someone does have any idea of what's going on or if i'm doing something wrong let me know ;)

McGiverGim, 2019-08-07:
I don't see any port selected. When you connect the FC it must appear in the list of ports and you only need to select it and connect. I don't see that you have selected it

ledevpole, 2019-08-08:
Hi!

When clicking on the flash firmware button it tries to connect to
"dev/000something.." but fail to open it when i ask for update. when
connecting for config it finds all alone the port.
I can send you a screenshot tonight if needed :P
Thank you for your quick response!

Yours sincerely

Paul

mikeller, 2019-08-08:
@ledevpole: It looks like the device drivers were not installed on your PC. Did you follow the instructions shown on the page when you start Betaflight configurator?

ledevpole, 2019-08-08:
Witch instructions? Nothing showed up. It asked me to download the
firmware, so i did it. it asked me to update my version of
betaflight-configurator so i did so.
 Do you mean that i have to download the drivers CP210XP and STM USB VCP
and ZADIG? If that so no i didn't. 😥
..

Docteh, 2019-08-09:
Are the CP210x Drivers just for F1 and F3 boards? I wonder if we can word that without being too confusing.

McGiverGim, 2019-08-09:
If we do it, maybe is better to divide the text into several little texts. Maybe something like text_X being X a number and the code only has to look in a loop until the message does not exist.

This text is very big, with a lot of HTML, and each change makes a headache to all the translators to let the HTML tags in place without forgot any.

mikeller, 2019-08-09:
@Docteh:

```
mikeller@pc:~/git/betaflight$ grep -L VCP src/main/target/*/target.mk
src/main/target/AIRHEROF3/target.mk
src/main/target/ALIENFLIGHTF1/target.mk
src/main/target/CJMCU/target.mk
src/main/target/ISHAPEDF3/target.mk
src/main/target/KISSFCV2F7/target.mk
src/main/target/MICROSCISKY/target.mk
src/main/target/MULTIFLITEPICO/target.mk
src/main/target/NAZE/target.mk
src/main/target/RACEBASE/target.mk
src/main/target/SPRACINGF3/target.mk
src/main/target/X_RACERSPI/target.mk
```

So amongst the supported boards in 4.1, there is only the KISSFCV2F7 left that does not use VCP (and that one is not included in our releases as it uses a custom boot loader).
I think it is time to rework the wording make it clear that the CP210x drivers are only needed for legacy targets and the KISSFCV2F7.

closed 2019-09-16 by stale[bot]