betaflight/betaflight-configurator #1572: Receiver tab is broken

atomgomba, 2019-08-10:
I'm using the Configurator on Ubuntu from the sources and cannot get the Receiver tab to work after commit 426e753b538103c94f8d92d228b24985d1850fd7. Channel values are not displayed and the bars aren't moving either. The things I've tried:

Complete rebuild:

```bash
rm -rf debug/ dist/ cache/ node_modules/
yarn install
yarn start
```

Complete rebuild and deleting contents of the local storage:

```bash
rm -rf ~/.config/betaflight-configurator
```

Only dropping back to commit 28d9bc36ecd71c2b1f9935d8ed03705e33220d95 fixes the problem. Though, I cannot see anything suspicious in the console. Below is a screenshot of what I see.

[image]

Docteh, 2019-08-10:
Hi, I've got a pending PR for this: https://github.com/betaflight/betaflight-configurator/pull/1570

Can you double check for me? if you scroll down to the bottom, and select a different refresh rate, the tab should start to update, no need for the patch.

#1570 moves the initial kickoff of updating the graphs to be below where they are defined.

atomgomba, 2019-08-25:
It is fine now. Thanks!

closed 2019-08-25 by atomgomba