betaflight/betaflight-configurator #1586: Led Overlay, Larson Scanner reversed direction label:"Feature Request"

m4R-T1n, 2019-08-18:
please can someone add an option to invert the running direction.
so you can adress 2 led strip next to each other running running mirrored.
like this :
[image]
thanks

mikeller, 2019-08-19:
@m4R-T1n: I really can't see why there would be need for this, if you can simply wire your strips the right way to make this work:

[image]

m4R-T1n, 2019-08-19:
not really, i can wire it like your drawing but then you have to rotate the led strip 180° to run the other way. but then the led stip has the mounting holes/ and chips mirrored.  
hope you understand the problem.
and if i wire like this booth strips will run starting from right to left

closed 2019-09-26 by stale[bot]