betaflight/betaflight-configurator #1592: Sections on VTX page not correctly aligned.

hydra, 2019-08-21:
[image]

I had a quick look at the HTML and it seems that the html is using a table layout inappropriately. 

It should be using the be using the `.cf_column` and related classes, refer to the gps.html / help.html / etc.

The two divs inside the first row have percentage widths which do not use 100% - there is one for 60% and one for 30%.  Likely all the the styles on `.tab-vtx .rightWrapper` can be removed.

The table should be removed, tr should be replaced by container div elements, td should be replaced by nested div elements inside the container div and the cf_column and related classes should be applied appropriately.

If you inspect the second row first td you can see how leftWrapper is being used incorrectly.

[image]

the .min-width 750px on the .vtx_table_box should not be needed when the vtx table is in a container that is 100% wide, i.e. this should be deleted:

```
.tab-vtx .vtx_table_box {
    min-width: 750px;
}
```

McGiverGim, 2019-08-21:
I'm not too sure about all but I will look into it when I have some time.

McGiverGim, 2019-08-28:
@hydra I've taken a look to this. It is not as easy, at least in the way it is now.

The 60% / 30% is a way to give it a little of room. I like to give some space if possible to the elements in the page. I think it looks better. I can use some margin or similar but this is another way to do it.

About the minimum for the table is necessary because it is a hack to allow the table be bigger than the left-wrapper when the window is at the minimum size. If you make it small you will see that the VTX Table can be bigger than the left-wrapper. It can be changed by a complete width wrapper, but I like the way it is aligned with the upper element when there is space enough.

I'm sure that all of this can be refactor in a better way, as always in CSS, but now I don't have too much time to fight against it so I will postpone it. If you have time, feel free to modify it. Maybe we need a document with a guide of how to do things, like the `cf_column` that you comment.

PR #1659 (2019-10-14): ISSUE-1592: Fix misaligned sections on VTX with modern flex. Remove gap between TDs in "current values" section
Fixed misaligned sections by removing table. Give "current values" section fixed width. Remove gap between TDs in "current values" section.
Fixes #1592
closed 2019-10-04 by stale[bot]