betaflight/betaflight-configurator #1593: Storage related exceptions on firmware flasher.

hydra, 2019-08-21:
With recent code and latest master as of today I'm seeing this stacktrace which needs investigation.  It happens when clicking on the firmware flasher button.

```
Error in response to storage.get: TypeError: $.get(...).error is not a function
    at Object.callback (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/jenkins_loader.js:107:16)
    at JenkinsLoader.loadBuilds (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/jenkins_loader.js:72:26)
    at Object.loader (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/firmware_flasher.js:246:50)
    at HTMLSelectElement.<anonymous> (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/firmware_flasher.js:309:46)
    at HTMLSelectElement.dispatch (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:42571)
    at HTMLSelectElement.v.handle (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:40572)
    at Object.trigger (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:70627)
    at HTMLSelectElement.<anonymous> (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:71222)
    at Function.each (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:2777)
    at k.fn.init.each (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:1419)
    at k.fn.init.trigger (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/node_modules/jquery/dist/jquery.min.js:2:71198)
    at Object.callback (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/firmware_flasher.js:573:62)
```

OS: Windows 10 x64.

McGiverGim, 2019-08-21:
@Docteh seems again https://github.com/betaflight/betaflight-configurator/pull/1568 ?

Docteh, 2019-08-21:
@McGiverGim I thought I got that in the follow up #1583

Docteh, 2019-08-21:
Yeah, line 107 is no longer `.error`
https://github.com/betaflight/betaflight-configurator/blob/c3a492f9b8ac7abc755ecb273040838bb8e76a6e/src/js/jenkins_loader.js#L107-L110

McGiverGim, 2019-08-21:
@hydra are you sure you are using the latest master?

mikeller, 2019-08-21:
@hydra: When reporting a bug against a development build, can you please post the changeset id that is giving you the bug? As we've discussed before, not doing so means you are wasting others' time by making them guess at what version is reproducing the problem.

hydra, 2019-08-22:
@McGiverGim yes, as of the exact time/date that i posted the issue.

@mikeller sure, will try and remember.

hydra, 2019-08-22:
finger slipped on mobile.... reopened

McGiverGim, 2019-08-22:
@hydra does not seem possible. Four days ago was committed a code that changed the error method for a fail method. Can you look at the line posted by @Docteh to see if you have a error (that appears in your stack trace) or a fail (the correct code)?

closed 2019-08-23 by no-response[bot]