betaflight/betaflight-configurator #1596: Use MSP_MULTIPLE_MSP in configurator for faster connections

awolf78, 2019-08-24:
With `MSP_MULTIPLE_MSP` you can retrieve multiple MSPs at once which greatly speeds up communication - versus a lot of ping pong if requesting and receiving MSPs slowed down by the BF scheduler. My Java script skills and knowledge about the configurator are probably not sufficient to implement this change myself, but should be easy enough for @McGiverGim ;)
If you request too many MSPs in the `MSP_MULTIPLE_MSP` request then it will only return as many as possible (depends on TX send buffer size for the corresponding serial/USB). You would then send another request to grab the remaining MSPs. Eventually when the older MCUs will get dropped within BF (or simply increase buffer sizes using #ifdef for larger MCUs) it will fit in the send buffer all at once ;)

McGiverGim, 2019-08-26:
It seems a good idea, but I can't see an easy way to do it if we don't know the buffer size. There is always a solution, but it will be a complicated wrapper about the actual code and I don't know if we will note some benefit. I will give it a test when I get some time, or maybe others can see a better solution.

awolf78, 2019-08-26:
Well that is the trick with `MSP_MULTIPLE_MSP`: You don't need to know the buffer size. It will simply return as many MSPs as it can fit in the TX buffer. On the client (configurator side) you would have to send another request for the missing ones. As the buffers on the BF side get larger you won't have to do this anymore.
I'm using `MSP_MULTIPLE_MSP` in ImpulseOSD and it has sped up things significantly for me.

mikeller, 2019-08-27:
@McGiverGim: I don't think that the additional effort required to make this work will be worthwhile for the one-off MSP requests when opening a new tab, but it might be attractive for the periodic update calls when connected, like the ones to get load, battery voltage, motor speed...

McGiverGim, 2019-08-28:
Yes, this will be a useful place. But I don't know how to integrate it in a clean way... the command will return a chunk of data, and they must be dumped in the MSP data structures that we use for each command (and they are unique for the app, not for the command). All of this having in mind that the actual system works in an asynchronous way, so other commands can ask for info at the same time. 

But I will play a little with it, perhaps I see a way to do it.

McGiverGim, 2019-08-29:
@awolf78 I'm trying to implement "as experimental feature" the MSP_MULTIPLE_MSP command, but looking at the firmware code I'm not too sure what data is sent and received.

If I'm not wrong, it sends a list of MSP codes as input, nothing more, as uint8 values.

What do I receive? The chunk of data for all commands one after other in the same chunk? Is there some prefix byte with the size of each command, some separator or similar? If this byte does not exist, we have a problem because we can have an outdated configurator with outdated MSP messages and we can process it wrong if some command has grown in size but the Configurator is not aware of that.

awolf78, 2019-08-29:
@McGiverGim 

1. You send a list of MSPs with no separator to BF as uint8_t values
2. You receive in the payload: the length of the corresponding MSP followed by the payload of the corresponding MSP
3. If the MSP does not exist it will return 0 as the length
4. If all the MSPs you requested do not fit in the TX buffer on the BF side it will only return the ones that fit.

mikeller, 2019-08-29:
@McGiverGim: Before you put too much more work into this now, I am not sure if now is a good time to look into making a fundamental change like this, considering that we are relatively close to a release - it's probably better to postpone this until after we've released configurator. This is even more so the case for the next configurator release as this will already bring Unified Target and VTX tables support, which are both crucial for Betaflight 4.1.

McGiverGim, 2019-08-29:
I'm only playing a little with it.. I think I will push a PR with the command added but without using it, for a future. And add this use after the release.

McGiverGim, 2019-08-29:
First version created, PR pushed... :)

awolf78, 2019-08-29:
Glad to hear you could make use of it @McGiverGim!

closed 2019-08-29 by awolf78