betaflight/betaflight-configurator #1601: Building a release with Node 11.X fails

serran0, 2019-08-27:
When running yarn install and/or yarn gulp release --win32 the following error started recently to occur:

error betaflight-configurator@10.6.0: The engine "node" is incompatible with this module. Expected version "10.x". Got "11.13.0"
error Found incompatible module
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
[00:14:55] 'dist_yarn' errored after 939 ms
[00:14:55] Error in plugin "gulpYarn"
Message:
    yarn exited with non-zero code 1.
Details:
    domainEmitter: [object Object]
    domainThrown: false

[00:14:55] 'release' errored after 5.31 s
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

Seems it should work if engine node in package.json was allowed above 10.x
Like this:

"engines": {
    "node": ">=10.x"
  }

serran0, 2019-08-27:
OK

closed 2019-08-27 by serran0

mikeller, 2019-08-27:
The tested node version for this 10. Just use this version.