betaflight/betaflight-configurator #1608: Video Transmitter tab shows "Not configured" when SA enabled on SS port

atomgomba, 2019-08-31:
On the Video transmitter tab I get the "Device not configured" notice when SmartAudio is enabled on soft serial. When I use a hardware UART it shows up okay.

McGiverGim, 2019-08-31:
I think this is a firmware issue. The Configurator shows the info that the MSP command returns.
@mikeller, can you move this to the firmware part?
@atomgomba can you attach a diff of the problem?

atomgomba, 2019-08-31:
Yes, so in this config UART3 is mapped to SERIAL_TX 12 and SmartAudio is enabled on SOFTSERIAL2:

```
# 

# diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.1.0 Aug 26 2019 / 20:58:05 (2b1d25639) MSP API: 1.42

# start the command batch
batch start

board_name OMNIBUSF4SD

# name: EX3

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A09
resource SERIAL_TX 12 B10

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -ARMING_GPS_NO_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
serial 31 2048 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 2 1 900 1100 0 0
aux 2 26 1 1900 2100 0 0
aux 3 35 3 1900 2100 0 0
aux 4 47 1 1400 1600 0 0
aux 5 49 3 1400 1600 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 2
vtxtable powervalues 0 1
vtxtable powerlabels 25 50

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = BIQUAD
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 100
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 250
set dyn_lpf_gyro_max_hz = 600
set acc_trim_pitch = 4
set acc_calibration = -1,91,-172
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 480
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 850
set vbat_max_cell_voltage = 420
set ibata_scale = 250
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_vbat_pos = 450
set osd_rssi_pos = 6413
set osd_link_quality_pos = 6412
set osd_tim_2_pos = 2518
set osd_flymode_pos = 14370
set osd_throttle_pos = 2505
set osd_vtx_channel_pos = 2091
set osd_current_pos = 2466
set osd_mah_drawn_pos = 419
set osd_craft_name_pos = 14393
set osd_avg_cell_voltage_pos = 14786
set osd_flip_arrow_pos = 14799
set osd_log_status_pos = 14375
set osd_stick_overlay_left_pos = 257
set osd_stick_overlay_right_pos = 278
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_profile = 3
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set vcd_video_system = PAL
set name = EX3

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 150
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass_type = BIQUAD
set dterm_lowpass2_type = BIQUAD
set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 15
set pidsum_limit = 1000
set throttle_boost = 0
set p_pitch = 38
set i_pitch = 85
set d_pitch = 35
set p_roll = 35
set i_roll = 78
set d_roll = 32
set d_min_roll = 16
set d_min_pitch = 18
set d_min_boost_gain = 30
set d_min_advance = 0
set ff_interpolate_sp = ON
set ff_spread = 18

rateprofile 0

# rateprofile 0
set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 95
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 76

# end the command batch
batch end

# 
```

mikeller, 2019-08-31:
@atomgomba: So just to make sure I understand you right: You are saying that when you configure SmartAudio on softserial with the configuration above, it is working, but it shows as 'Device not configured' in the Video Transmitter tab?

@McGiverGim: Not yet sure if this is firmware or configurator, leaving it here for now.

McGiverGim, 2019-09-01:
Maybe I must start with the basic. @atomgomba the SmartAudio is working in softserial out of the Configurator? You can change channels with the OSD, lua or CLI commands?

atomgomba, 2019-09-01:
No, I cannot get this Unify Pro Nano to work with my Omnibus F4 V3. I accidentally dropped this VTX into sea water, but it is working great as a VTX ever since. I'm not sure if possibly only the circuitry for Smart Audio got damaged, but this is not the point here. 
The point is when I select UART3 as port for SA, I get the Video transmitter tab fully displayed, including a status box that says "Device ready: no". It is totally expected as I couldn't get the device to work so far. But when I map UART3 to SOFTSERIAL2 and select SOFTSERIAL2 as port for SA, the Video transmitter does not display, apart from one or two notices and one of them says "Device not configured".
What I would expect is that Video transmitter tab behaves the same in both cases, regardless of the device is actually working or not, since the only difference is the port on which Smart Audio is activated.

McGiverGim, 2019-09-01:
@etracer65 did you some test with softserial? 

The message that you say appears when you not configure any VTX device. It does not matter if you have it connected or not.

Another thing is the device ready, it needs the device connected and powered if I'm not wrong.

etracer65, 2019-09-01:
No, I don't have any rigs set up with softserial and SA.

McGiverGim, 2019-09-01:
Yes, but for example simply enabling a TX as SmartAudio, without any VTX device connected, the VTX Table appears because the MSP returns a SmartAudio device. It seems that with SoftSerial this does not happen. I'm out of home so I can test it now.

McGiverGim, 2019-09-02:
@etracer65 I have tried to configure it in a SoftSerial on an OMNIBUSF4SD, with this result: 
- Using as softserial a TX: it work
- Using as softserial another pin (C8 in this case): the MSP command does not detect the SmartAudio configuration. Maybe the problem is the pin, that is not supported for that, but the SoftSerial port appears in the port tab and I can assign the SmartAudio to it.

mikeller, 2019-09-02:
Ok, I found one case where I can reproduce it as not working:
If resources are configured as softserial 2, but none as softserial 1, then softserial 2 appears in the ports list, and can be assigned to SmartAudio, but the VTX tab does not work I have not tested if this will work to control a VTX, but it is broken anyway, softserial 2 should either not show up at all (like what happens for say motor 6 if motor 5 does not have a resource configured), or it should fully work.

atomgomba, 2019-09-02:
@mikeller OMNIBUSF4SD has three UART's, UART1, UART3 and UART6. Only UART6 has inverter, so I'm using it for SBUS. I have UART1 (aka. SOFTSERIAL1) configured for s.port telemetry, thus I wanted to use UART3 (aka. SOFTSERIAL2) for SmartAudio. So in my case both soft serials were set.

mikeller, 2019-09-02:
@atomgomba: I did not mean to imply that my reproduction was what you are getting - when I try that then everything seems to be working for me. I just supplied the information that I found to help narrow this down.

McGiverGim, 2019-09-02:
@atomgomba I tested your diff and it seems to work for me. I can see the VTX tab with the VTX Table without problem. 

It appears the `Device Ready: No` but I don't have any VTX connected.

What versions of firmware and Configurator are you using? Latest master on both?

atomgomba, 2019-09-02:
@McGiverGim A very recent CI build. It is shown in the diff above: Aug 26 2019 / 20:58:05 (2b1d25639). And I'm running the Configurator from git master using yarn.

atomgomba, 2019-09-02:
@McGiverGim I've just noticed my diff was wrong. In it there is SmartAudio enabled on both UART3 and SOFTSERIAL2. Sorry, it is my fault! You have to uncheck SA for UART3 on Ports tab and have it enabled only on SOFTSERIAL2. I could reproduce it in this moment using same version of firmware and latest Configurator master.

mikeller, 2019-09-02:
@atomgomba: Yes, that is expected behaviour - we do not catch / fix the problem if a VTX protocol is configured twice, but it won't work.

atomgomba, 2019-09-02:
@mikeller Oh yes, what I tried to express was that @McGiverGim couldn't reproduce the error in question because of my badly prepared diff. It seems to me it's Monday.

McGiverGim, 2019-09-03:
Yes, exactly. With the "new" diff the result is the same that I had before: it seems it does not detect if SmartAudio is present at some SoftSerial port. 
I suppose the firmware only searchs for it at standard UART ports at least in the data returned in the MSP command. To me it seems a bug of the firmware, but @mikeller and @etracer65 must confirm that.

etracer65, 2019-09-03:
The `deviceIsReady` flag was already in the `MSP_VTX_CONFIG` message - it wasn't added for `vtxtable` support. So there may be some deeper problem with:
```
deviceIsReady = vtxCommonDeviceIsReady(vtxDevice)
```
in regards to vtx control configured on softserial.

etracer65, 2019-09-03:
Or deeper with the `vtxDevice` not being configured.
```
            const vtxDevice_t *vtxDevice = vtxCommonDevice();
            unsigned vtxStatus;
            vtxDevType_e vtxType = VTXDEV_UNKNOWN;
            uint8_t deviceIsReady = 0;
            if (vtxDevice) {
                vtxCommonGetStatus(vtxDevice, &vtxStatus);
                vtxType = vtxCommonGetDeviceType(vtxDevice);
                deviceIsReady = vtxCommonDeviceIsReady(vtxDevice) ? 1 : 0;
            }
```

McGiverGim, 2019-09-03:
Maybe I have explained it too bad... sorry... the problem is not the `deviceIsReady` flag, is the `vtx_type`, that is returning Unsupported or Unknown, in place of returning SmartAudio, when the SmartAudio is configured in a SoftSerial port.

It seems to have a problem too with the `deviceIsReady` but this another problem.

atomgomba, 2019-09-03:
I'm a total noob when it comes to the firmware code, but `vtxCommonSetDevice(&vtxSmartAudio);` is called here in [vtx_smartaudio.c](https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c#L720) and before that is a condition for the serial port:

```c
    if (!smartAudioSerialPort) {
        return false;
    }
```

Is it possible that `smartAudioSerialPort` is not allowed to be a soft serial port? Sorry again if this sounds like nonsense.

McGiverGim, 2019-09-03:
I can confirm that MSP sends a 255 (Unknown) and not 3 (SmartAudio) when the SmartAudio is configured over SoftSerial, at least with the diff provided.