betaflight/betaflight-configurator #1609: MSP RX control sticks window does not reflect dark mode setting

jflyper, 2019-09-01:
[image]

IvoFPV, 2019-09-01:
I'll take a look.

Anisotropic, 2019-09-16:
"Control sticks" is absent in my bf-configurator. How can I force to show this button? The receiver is powered on my connected quad, and taranis is binded and also powered.

Anisotropic, 2019-09-16:
@IvoFPV I found solution how to pass darkTheme variable to separate window. Now working on it.

McGiverGim, 2019-09-17:
> "Control sticks" is absent in my bf-configurator. How can I force to show this button? The receiver is powered on my connected quad, and taranis is binded and also powered.

You must select MSP as RX protocol.

PR #1670 (2019-10-14): ISSUE-1609: Create communication mechanism between windows, implement…
ISSUE-1609: Create communication mechanism between windows, implement dark theme fix for sticks window using it. Small fix for Tip popup window.
Fixes #1609 
[image]
[image]
[image]
closed 2019-10-14 by mikeller