betaflight/betaflight-configurator #1620: Add Degree field to ACC sensor page

Ozzy33, 2019-09-04:
Would be nice to add accelerometer degrees to the sensor page (next to the G's numbers), like -90 to 0 to +90 degrees. Also add a calibrate ACC button on this page (like the setup tab has).
[image]

mikeller, 2019-09-05:
@Ozzy33: Ok, I can understand the 'degrees' output. But I am not sure about the 'calibrate ACC' button - having this in one place seems to be enough.

Ozzy33, 2019-09-06:
Hi Mike, that would be fine! 
It would be nice to have 0.1 degree resolution if the numbers are not jumping all over (filtered data), if its moving around to much, then 1 degree would be fine too.
Thanks.

Ozzy33, 2020-10-02:
Any updates on this feature? TIA!

haslinghuis, 2020-10-04:
working on this:

See https://forums.parallax.com/discussion/114274/tri-axis-g-force-to-radius-and-degree-conversion-question

Added some formulas but it only settles if acc is stable

Ozzy33, 2020-10-05:
Maybe a averaging filter?
And thanks for the update!

PR #2197 (2020-10-08): Add degrees to accelerometer for X and Y axis in sensor tab
Fixes #1620
[image]
closed 2020-10-08 by mikeller