betaflight/betaflight-configurator #1630: CLI AutoComplete missing some options

McGiverGim, 2019-09-07:
Using it, I usually use the command `diff all defaults`, one of the most useful of them for me... the automplete suggest `diff` but it does not suggest the `all` parameter (and of course it does not suggest the `defaults` at the end).

@Cleric-K can this be fixed or the FC does not give enough information? If it does not provide them, maybe it must be added to the info of the command in the FC.

Docteh, 2019-09-08:
Looks like the FC does give enough information.
4.0.3
```
dump - dump configuration
	[master|profile|rates|hardware|all] {defaults|bare}
```

4.1.0
```
diff - list configuration changes from default
	[master|profile|rates|hardware|all] {defaults|bare}
```

Regex is dark dark magic.

DHaacke, 2022-02-11:
Just a comment, but from what I've noticed after a cursory glance through the code is that cli commands aren't missing per se, but likely not being added to the autocomplete value table because its hardware category isn't active or selected. I came across what I thought was a missing value when trying to set an i2c1_clockspeed_khz value. I wasn't getting the standard cli reply. Later I realized I hadn't enabled the magnetometer, and once that was done, the command showed up. Something tells me bad things would happen if users could start changing values for non-active entities. The alternative would be to always display everything, but non-entities would be marked with an asterisk (or greyed out an not selectable) to denote the option is there but can't or shouldn't be set. Just thinking out loud...

haslinghuis, 2023-06-21:
Fixed in: https://github.com/betaflight/betaflight-configurator/pull/1677

closed 2023-06-21 by haslinghuis