betaflight/betaflight-configurator #1632: Debug mode not set correct in Blackbox tab

HrrrPippi, 2019-09-07:
Hi there
When setting Dshot_rpm_telemetry via the blackbox tab in configurator, it reads dyn_lpf in the cli
When setting it to rpm_filter it reads RX_SPREKTRUM_SPI in the cli
and d_min is set to dshot_rpm_telemetry in the cli.

Seems like the entire list is pushed in some direction.

Just testing configurator 10.6.0, with the new sliders added to the pid tuning tab(dont know which version)

IvoFPV, 2019-09-07:
Are you on the latest version of firmware?

IvoFPV, 2019-09-07:
I just tested it on the latest builds and I can't replicate the issue. It is probably your firmware version before dual gyro PR where some debug modes were removed. Please update and if you still have issues provide a `diff`.

HrrrPippi, 2019-09-07:
Cool... Thanks..

closed 2019-09-07 by HrrrPippi